OrcaSlicer/OrcaSlicer #8326: Upload and print feature does not capture filament change custom gcode

Griffinator8r, 2025-02-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

AMD hardware if that matters...

### Printer

Voron 2.4, ERCF V2

### How to reproduce

1. Edit the filament change gcode
2. Use the export gcode after slicing, everything works as expected, I get the commands I want on filament/tool change
3. Use the 'upload/print' feature, my gcode overrides no longer show up, and I get the default MMU tool change commands as though I did not edit anything

### Actual results

exporting a gcode file vs uploading and printing has different results for the machine gcode for filament change gcode - I have not tried putting other overrides in to test out the other gcode overrides for differences in behavior

### Expected results

upload and print should output the same gcode as exporting to a file

### Project file & Debug log uploads

[demo.zip](https://github.com/user-attachments/files/18683468/demo.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

Not sure if this is expected behavior or not, but I would have expected these two commands/options to produce the same gcode outputs and treat everything the same, if that is not expected then maybe make that clear somehow?

Griffinator8r, 2025-02-06:
[ExampleFiles.zip](https://github.com/user-attachments/files/18683516/ExampleFiles.zip)
Adding example files, labeled appropriately showing the difference between an exported gcode file and a upload and printed file

Noisyfox, 2025-02-09:
> In case it helps, a friend pointed out that the uploaded file has "; processed by HappyHare" at the top, so there appears to be post processing on the files that get uploaded as opposed to exported...?

Yeah that's HappyHare on your machine that post process the gcode when the print starts.

Griffinator8r, 2025-02-09:
Ok, good to know I guess then, so probably not an orca slicer issue then, I'll close this, I seem to be the only one that has a different, unrelated issue that the MMU commands behave differently than the T0 commands for some reason, that's the only reason I came across this behavior.

closed 2025-02-09 by Griffinator8r