OrcaSlicer/OrcaSlicer #8328: Bed Type stuck on Smooth High Temp Plate.  Started in version 2.2.0

steadmanc, 2025-02-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 and 2.2.0

### Operating System (OS)

Windows

### OS Version

11

### Printer

Flashforge Adventurer 5M Pro

### How to reproduce

NA since it in the left pane on startup.

### Actual results

[image]

### Expected results

Should be able to change the Bed Type via drop down but it's locked.

### Project file & Debug log uploads

Version 2.2.0 is when it started.  Confirmed it is still in 2.3.0 nightly build.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

discip, 2025-02-06:
Thing is: If you share a project file all your settings are comprehensible.

It's not about the part rather than about your settings. 😊

Just put a generic cube on the plate.

igiannakas, 2025-02-06:
Enable multi plate support in the printer settings.

[image]

steadmanc, 2025-02-06:
Bingo!  Apparently, those were enabled when I installed the software but had since been turned off when installing the updates.  Exactly the help I needed.  Thank you.  Pics with the check boxes I had to check to get those features turned back on.  

 

Thank you.

 

image.jpeg (view on web) <https://github.com/user-attachments/assets/6a2f2d06-bec5-4cb2-8f3c-5ed228f46c79>

closed 2025-02-06 by steadmanc