OrcaSlicer/OrcaSlicer #8330: pause print for short first layer times

Saibot-ux, 2025-02-06:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I want to have a high bed temperature in the first layer and try to cooldown quickly for the next layers. thats not possible for small objects. 

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

I would be happy if there is a feature where the printer pauses if the layer time or the base area of the objects is small. So the bed has time to cooldown to a certain temperature.

discip, 2025-02-06:
@Saibot-ux
In your start g-code you just need to use **`M190 R...`** instead **`M140 S...`**.

e.g.:
before
**`M140 S100`**

after
**`M190 R100`**

Here is the reference in case you want to know more: [Marlin G-Codes](https://marlinfw.org/docs/gcode/M190.html)

discip, 2025-02-06:
@Saibot-ux
In your start g-code you just need to use **`M190 R...`** instead of **`M140 S...`**.

e.g.:
before
**`M140 S100`**

after
**`M190 R100`**

Here is the reference in case you want to know more: [Marlin G-Codes](https://marlinfw.org/docs/gcode/M190.html)

ShaneDelmore, 2025-02-06:
It wouldn't be start g-code here though, it would be gcode for layer 1 change.  Can you explain the benefit of this feature?