OrcaSlicer/OrcaSlicer #8331: Auto leveling between prints is not being saved

steadmanc, 2025-02-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 and 2.2.0

### Operating System (OS)

Windows

### OS Version

11

### Printer

Flashforge Adventurer 5M Pro

### How to reproduce

NA since it is during the print.

### Actual results

I was able to level the Flashforge 5M Pro once upon setting up the printer.  I was then able to print different items,etc, without having to relevel the plate.  As of 2 versions ago, I found I have to relevel before each print or the print go very badly.  While trying to troubleshoot the current Bed Type and Filament bugs, I ran a print with FlashPrint and it worked find.  This leads me to believe the problem is when the print starts with the OrcaSlicer.  It seems like it may not be using the previous print session leveling data or may be skewing it somehow before printing.  The work around with Orca-Flash is making it level each time.  I have no idea how to make OrcaSlicer use the leveling feature for FlashForge 5M Pro.

### Expected results

Should be able to add or remove filaments in the left pane.  

### Project file & Debug log uploads

Version 2.2.0 is when I first noticed it in Orc Slicer.  Confirmed it is still in 2.3.0 nightly build.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DVSVIDEO, 2025-02-06:
Levelling is the job of the printer and its firmware. 
It has nothing to do with the slicer software.
 It has to do with the leveling, in that you can request the execution of the leveling commands (which is done by the printer firmware) in the start script section of the slicer. Why don't you ask for a solution in the forums. 
This is not a bug. Please close this post.

steadmanc, 2025-02-06:
The Orca-Flashforge allows selecting leveling in the software.  It may be how Flashforge integrated that feature with the latest Orcaslicer versions.

closed 2025-02-06 by steadmanc