OrcaSlicer/OrcaSlicer #8335: Second layer of print skips ~50% of the gcode

havedill, 2025-02-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

5800X3D
3080TI

### Printer

Anker M5C 0.4 Nozzle

### How to reproduce

I've seen this once before, way back in the past. I think maybe even for this same STL.

The second layer SKIPS 50% of the gcode while laying down the bottom shell layers. Then it continues on fine for each layer after..

### Actual results

[image]

[image]

### Expected results

Should be a solid layer like the gcode preview shows

### Project file & Debug log uploads

[PLANTER profile.zip](https://github.com/user-attachments/files/18695153/PLANTER.profile.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

havedill, 2025-02-07:
I tried "fixing stl" and got slightly different results. With two new lines in the space where the gcode is skipped before it continues onto the 3rd layer

[image]

havedill, 2025-02-07:
It's a simple model, not made by me originally. So i recreated my own STL and the issue went away. Just odd Orcaslicer did not show the weird print issues in the gcode

But it must be something to do with that STL. Here's my recreation which seems to be currently printing fine:

[Base_Remade.zip](https://github.com/user-attachments/files/18705490/Base_Remade.zip)

Ergonomicmike, 2025-02-09:
@havedill Is this still an Issue? Or simply a problem with the orginal? If no longer an Issue it would be good to close this Issue so that the devs don't spend their time reading about a non-Issue.

havedill, 2025-02-09:
I would say it's still an issue. The slicer shows incorrect gcode previews

explosive-gerbil, 2025-04-09:
I've experienced this with various models, especially after moving to the 2.3.0 versions (including RC builds). I am currently using build 70931e5.