OrcaSlicer/OrcaSlicer #8336: Rounded Walls much worse on Orca Slicer compared to Bambu Slicer

randyre, 2025-02-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Win 10

### Additional system information

Orca slicer profile is set to same parameters as Bambu Slicer. Same object sliced and printed. Orca Sliced versions outer walls are rough and jagged, Bambu is much smoother, only difference is the slicer used. What gives?
One more interesting note is as I watch the Orca Slicer version print the outer wall of the initial layer is near perfectly round, its only when it starts to print the second lay that the faceted edges happen.

### Printer

Bambu Labs X1C, P1S and Creality K1 Max

### How to reproduce

1. Duplicate slicer setting from known good in bambu Slicer to orca Slicer
2. Slice same object with rounded walls
3. Print

### Actual results

Orca slicer walls are faceted and jagged

### Expected results

Should be as smooth as Bambu Slicer

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Kinofaya, 2025-02-06:
Do you use the same file type for your 2 objects ?
It looks like you imported a high poly ;stl file or a .STEP file to BambuStudio, and maybe from there exported the object file as a .3mf and lost some resolution ?

randyre, 2025-02-06:
Same exact file for both slicers. it's a standard Tinkercad STL file

randyre, 2025-02-06:
> Do you use the same file type for your 2 objects ? It looks like you imported a high poly ;stl file or a .STEP file to BambuStudio, and maybe from there exported the object file as a .3mf and lost some resolution ?

Same exact file for both slicers. it's a standard Tinkercad STL file

Noisyfox, 2025-02-07:
Could you provide 3mf files from both BBS and Orca that have everything set to the same?

randyre, 2025-02-07:
> Could you provide 3mf files from both BBS and Orca that have everything set to the same?

Sure can.

closed 2025-05-17 by github-actions[bot]