OrcaSlicer/OrcaSlicer #8337: Running on Linux using flatpak, window content is black

nuxeh, 2025-02-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

V2.3.0-dev

### Operating System (OS)

Linux

### OS Version

NixOS

### Printer

n/a

### How to reproduce

1. Obtain `OrcaSlicer-Linux-flatpak_V2.3.0-dev.flatpak` from github releases page
2. run `flatpak --user install ~/Downloads/OrcaSlicer-Linux-flatpak_V2.3.0-dev.flatpak`
3. run `flatpak --user run io.github.softfever.OrcaSlicer`

### Actual results

[image]

### Expected results

Program runs correctly

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Athoskal, 2025-02-07:
### Work around

Switch to x11. 🤢

Getting this as well on flatpak, and appimage. It seems to be related to Wayland/XWayland.
Don't know if this helps but I had the same issue on windows 10 until I let microsoftedgeupdate though a firewall.

### OrcaSlicer Version

V2.3.0-dev

### Operating System (OS)

Linux

### OS Version

Arch, KDE Plasma 6.2.5
### Additional system information

Kernel Version: 6.13.1-arch1-1 (64-bit)
Graphics Platform: Wayland
Graphics Processor: NVIDIA GeForce RTX 30XX/PCIe/SSE2

Noisyfox, 2025-02-08:
`export WEBKIT_DISABLE_COMPOSITING_MODE=1` then run the app.

nuxeh, 2025-02-08:
> `export WEBKIT_DISABLE_COMPOSITING_MODE=1` then run the app.

Incredible, that seems to have done the trick. It now works and displays the window contents. Thanks!

nuxeh, 2025-02-08:
Can confirm that after the above fix, the flatpak is fully functional, I've tested it quite thoroughly in actual use now

Athoskal, 2025-02-09:
Glad it seems to be working for some! still broken on both flatpak and appimage for me. I at least can see the startup wizard now after adding  WEBKIT_DISABLE_COMPOSITING_MODE=1 but the main preview still does not render.

witc-git, 2025-02-23:
> Glad it seems to be working for some! still broken on both flatpak and appimage for me. I at least can see the startup wizard now after adding WEBKIT_DISABLE_COMPOSITING_MODE=1 but the main preview still does not render.

Try #8372 in addition to the webkit disable stuff mentioned above, worked for me for a while.

etrigan63, 2025-02-24:
I have encountered this same issue with the beta flatpak on my workstation running CachyOS (Arch) and Nvidia graphics. On my laptop (Framework 13 AMD with AMD Graphics) the flatpak behaves normally. This leads me to believe that Nvidia is to blame here. At least for me.

etrigan63, 2025-02-26:
2.3.0-beta2 flatpak updated this morning and still no change. Work area fails to show the build plate.
CachyOS Linux (Arch) + Nvidia RTX A2000 (running 570 series open source drivers). The application is running under XWayland so the `zink` patches have not shown up. Trying to force it to run on Wayland via `flatseal` does nothing. It still runs in XWayland.

Noisyfox, 2025-02-27:
> 2.3.0-beta2 flatpak updated this morning and still no change. Work area fails to show the build plate. CachyOS Linux (Arch) + Nvidia RTX A2000 (running 570 series open source drivers). The [...]

Maybe try adding `ZINK_FORCE_OVERRIDE=1`?

etrigan63, 2025-02-27:
I reinstalled the flatpak and added the zink env variable. No change and it still launches XWayland. Do the other env variables need to be there too?

etrigan63, 2025-02-27:
Also, when I connect the printer profile to Klipper, `orca-slicer` crashes.

Noisyfox, 2025-02-27:
> Also, when I connect the printer profile to Klipper, `orca-slicer` crashes.

See https://github.com/SoftFever/OrcaSlicer/issues/7210

Athoskal, 2025-02-27:
Getting a similar problem on EndevourOS (EOS is just arch with GUI installer it uses arch repos/kernels and updates the same as arch)  I tried to force zlink as well. 50_mesa.json is where it should be. The old overrides never worked for me either.

Operating System: EndeavourOS 
KDE Plasma Version: 6.3.1
KDE Frameworks Version: 6.11.0
Qt Version: 6.8.2
Kernel Version: 6.13.4-arch1-1 (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 5800X 8-Core Processor
Memory: 62.7 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 3080
NVIDIA Driver 570.86.16
using nvidia-utils

klylabs, 2025-03-01:
> > 2.3.0-beta2 flatpak updated this morning and still no change. Work area fails to show the build plate. CachyOS Linux (Arch) + Nvidia RTX A2000 (running 570 series open source drivers). The application is running under XWayland so the `zink` patches have not shown up. Trying to force it to run on Wayland via `flatseal` does nothing. It still runs in XWayland.
> 
> Maybe try adding `ZINK_FORCE_OVERRIDE=1`? Also does `/usr/share/glvnd/egl_vendor.d/50_mesa.json` file exist on your distro since there are cases that this file is on a different path.

For arch based users facing this issue, they would need to download the latest nightly builds and run the AppImage directly, since the aur versions only point to the beta version and the patchdidnt get merged in the beta build, 
building it by just changing the aur pkgbuild to point to the nightly builds also wont work,they will need to modify the script to have it use the AppRun and not remove it

I'll create an aur account to make a pull request to fix this once i have time

@Athoskal @etrigan63

Athoskal, 2025-03-02:
> 
> For arch based users facing this issue, they would need to download the latest nightly builds and run the AppImage directly, since the aur versions only point to the beta version and the patchdidnt get merged in the beta build, in addition to that building it by just changing the aur pkgbuild to point to the nightly builds also wont work since the maintainer for that package deletes AppRun and just directly runs the app, they will need to modify the script to have it use the AppRun and not remove it
> 
> I'll create an aur account to make a pull request to fix this once i have time
> 
> [@Athoskal](https://github.com/Athoskal) [@etrigan63](https://github.com/etrigan63)

I had the issue with the nightly flatpak I have not used the version in the aur

agusmor, 2025-03-18:
Similar problem here, Fedora 41 running KDE Plasma (wayland) on gtx1650 with driver v570.
Appimage v2.3.0-rc works fine. Flatpak does not render bed/model, even with all the environment variable overrides posted here.

Athoskal, 2025-03-18:
can confirm it works fine on appimage now

Indomitable, 2025-04-02:
Hi, Today noticed that the AppImage support is removed, for me only it was working. 
  Fortunately I've able to find a way to work. My suspicion was that there flatpak prevents OrcaSlicer to access the 50_mesa.json file located in the `/usr` folder so we can point the EGL_VENDOR to the home folder use 
 `flatpak override --user io.github.softfever.OrcaSlicer --env=__EGL_VENDOR_LIBRARY_FILENAMES=$HOME/.local/share/flatpak/runtime/org.freedesktop.Platform.GL.default/x86_64/24.08/active/files/glvnd/egl_vendor.d/50_mesa.json` also add access to the file system `flatpak override --user io.github.softfever.OrcaSlicer --filesystem=$HOME/.local/share/flatpak/runtime/org.freedesktop.Platform.GL.default/x86_64/24.08/active/files/glvnd/egl_vendor.d`
I hope it helps.

Indomitable, 2025-04-04:
You may go without installing the package. Copy the 50_mesa.json file from `/usr/share/glvnd/egl_vendor.d/50_mesa.json` to the home folder and point `__EGL_VENDOR_LIBRARY_FILENAMES` to it. The problem is that flatpak has no access to it.

Blogshot, 2025-04-05:
> You may go without installing the package. Copy the 50_mesa.json file from `/usr/share/glvnd/egl_vendor.d/50_mesa.json` to the home folder and point `__EGL_VENDOR_LIBRARY_FILENAMES` to it. The problem is that flatpak has no access to it.

This worked for me.

For the lazy peple:
```
sudo cp /usr/share/glvnd/egl_vendor.d/50_mesa.json ~/
flatpak override --user io.github.softfever.OrcaSlicer --env=__EGL_VENDOR_LIBRARY_FILENAMES=$HOME/50_mesa.json
```

closed 2025-07-12 by github-actions[bot]