OrcaSlicer/OrcaSlicer #8338: Enable the use of LAN connected printers in the Multi-device feature

DanielEikelenboom, 2025-02-07:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

When sending a print job to multiple printers I:
- need to press through all the send print sequence everytime
- set all the AMS settings for every print (even though they are all the same)
- need to wait between every send until it has been send through LAN

Multi-Device setting has a perfect way to mitigate this, only it:
- Works only with Bambu cloud
- Can only add 6 printers

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Enable the use of LAN connected printers in the Multi-device feature.

(secondary, allow for more than 6 printers)

### Describe alternatives you've considered

1. Write a script to send all the files at once over FTP
2. Paid alternative farm management software

### Additional context

The slicer has access to a lot of printers. As can be seen, the Mini is put on 'Cloud'-mode. The rest is LAN only.
[image]

When adding printers to the Multi-Device tab, only the cloud printer can be added.
[image]

Additional benefits:
- All printer statusses are viewable from 1 window
- Update all print files in 1 click
- Use the feature to start all prints in 1 click, but delay the heating of the bed (to not blow the fuses)
- Could be useful for different brands of printers if done without Bambu cloud

KevinH300, 2025-02-08:
This would be a great feature, especially given the recent announcements regarding Bambulab's "security measures"

berezovskyi-oleksandr, 2025-02-11:
I switched all my printers to the LAN mode after the Bambulab's "security updates". Would love using this feature again

nate1mace, 2025-05-06:
Bumping - The fastest way to send prints used to be via Handy app - but now that I'm on LAN every print takes 2 mins to set up x 15 printers x 2 day runs it's an hour a day just spent doing something you in theory can hit one button for. Any thing that shaves just seconds off the print process imo are critical - but this would shave hours.

Sawtaytoes, 2025-07-02:
I'm in the same boat. I just got into 3D Printing recently, but I have 3 of the X1 Carbon printers, and I have to manually set each one up. It'd be great if I could send my build plates to all 3 all at the same time.

Sarkasaa, 2025-07-05:
Same here, I started out with an A1 Mini + AMS lite and recently bought an A1. And switching around differently-sized plates and their profiles before issuing a print is pretty tiresome.

mara5a, 2025-07-07:
same here, multi-device with LAN connected printers is something that'd be greatly appreciated.