OrcaSlicer/OrcaSlicer #8346: General Issues with Tree Hybrid supports

Surpixx, 2025-02-08:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0 and 2.3-dev

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

Select Tree hybrid support (doesn´t matter if manual or auto)

Slice your Object

### Actual results

1. Issue:  Tree hybrid supports only really work if the option "on build Plate only" is turned off as mentioned in issue  #6618, that has been marked as stale. This however is an option that should be working.

2. Issue: False amount of top support interface layers. There is always one layer missing, regardless of the set value with the exception of 0, then it correctly slices the object with the right amount of support layers with no gap.

3. Issue: No option to customize the Support interface with the build plate when using tree hybrid support. This has also been mentioned in issue   #6635. There is no option to set the infill density of the tree supports or to change the base pattern.

4. Issue: Artifacts in the support regions. When assigning "normal" supports for an area, it sometime generates single lines of support in the air as shown in the picture below

[image]

[image]

### Expected results

The mentioned issues shouldn't happen/ customization should be implemented.

### Project file & Debug log uploads

[Bottom Shell - Kopie.zip](https://github.com/user-attachments/files/18718034/Bottom.Shell.-.Kopie.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

twoscoopsofpig, 2025-07-19:
I'm having issues with this as well, though I can't verify that the missing layer issue sis happening as it simply refuses to finish slicing with hybrid tree supports if I limit the supports to on build plate only. It just hangs at 69% and I have to force quit to regain control.

Has there been any movement on this issue?

roxorplox, 2025-07-21:
I am also having issues with this. When using Tree Hybrid, it wont generate grid supports, and uses only trees.

roxorplox, 2025-07-21:
Can we get someone assigned to this bug to be fixed for 2.4.0. Hybrid Tree Supports are crucial. This issue in particular is plaguing me currently.

    Issue: Tree hybrid supports only really work if the option "on build Plate only" is turned off as mentioned in issue 

https://github.com/SoftFever/OrcaSlicer/issues/6618, that has been marked as stale. This however is an option that should be working.

AssiveMass, 2025-11-26:
These issues make the Tree Hybrid useless. The concentric pattern that starts in the middle https://github.com/OrcaSlicer/OrcaSlicer/issues/6635 ruins most prints. Even when it sticks it looks like it is over-extruding like hell, when rest of the first layer is perfect.