OrcaSlicer/OrcaSlicer #8351: Please Implement Mirror Paint Tool

NicoIic, 2025-02-09:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Myself alongside other people struggle w/coloring details on models symmetrically. This issue also expands to artists who digitally sculpt their own models, this results in their work having unsymmetrical details. Attempting to color symmetrically, but lacking a proper tool additionally takes a long time to produce w/less-than-ideal results. After much consideration, I realized this was a much more widespread issue for people that justified moving this issue from "Ideas" to here. 

### Which printers will be beneficial to this feature?

Others, All

### Describe the solution you'd like

If feasible, I'd love to eventually see a mirror paint tool implemented on OrcaSlicer. It'll essentially be an adjustable line that best centers itself on the model (approximately if the model isn't perfectly symmetrical). When you color one half of the model, it'll automatically duplicate itself on the other. I've attached an image below of a drawing app w/a similar feature. Thank you for reading this!

[image]

### Describe alternatives you've considered

1. I've attempted to color details on one side of a model > duplicate a model > flip the new model onto the opposite side > center both models > assemble them together. This seemingly would work, but it sadly produced a lot of poor results (e.g. jagged lines, etc.) once sliced. 
2. I've additionally tried modeling and sculpting programs such as Nomad Sculpt. They have a mirror paint tool, however it also sadly produced a lot of poor results (e.g. blurry details when they originally weren't, etc.) once imported onto OrcaSlicer.

ShaneDelmore, 2025-02-09:
If the model was perfectly symmetrical you might try cutting it in half, painting the remaining half, then mirroring it and putting it back together as a temporary workaround.

NicoIic, 2025-02-09:
> If the model was perfectly symmetrical you might try cutting it in half, painting the remaining half, then mirroring it and putting it back together as a temporary workaround.

Thank you for sharing! I've heard of this method before, but it sadly isn't perfect as people are likely to put the model back incorrectly (etc). However, I fully agree it'll work as a temporary workaround in certain cases. I'm hoping a proper tool is a feasible request as it'll be more ideal and greatly appreciated by many in the long run.