OrcaSlicer/OrcaSlicer #8352: Command Line Interface

InputBlackBoxOutput, 2025-02-09:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

🔍I am interested in performing a grid search for optimal print configurations in the OrcaSlicer using a CLI interface.

🥲I cannot find a command line interface for the tool. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

A command line interface which accepts different configurations and then reports slicer stats

🚀I would love to work on adding this feature if the team provides me directions to get started

### Describe alternatives you've considered

Bambu studio currently supports CLI usage

InputBlackBoxOutput, 2025-02-09:
Related to issue: https://github.com/SoftFever/OrcaSlicer/discussions/1603

igiannakas, 2025-02-09:
The print stats are available as variables in orca, could use a standard cout command to print them to the programme output? Or write them to a file?
 
They are available in the gcodeviewer.cpp file but I’m unsure if this gets invoked when slicing from the command line