OrcaSlicer/OrcaSlicer #8360: [Nightly] Prusa Mk4s profiles not showing up in system printers list after they were added

filloax, 2025-02-09:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

Nightly build [107cd3](https://github.com/SoftFever/OrcaSlicer/commit/107cd3edb05ef7327d6925aa352082f1168473e0)

### Operating System (OS)

Windows

### OS Version

11

### Printer

Prusa Mk4s

### How to reproduce

1. Open add/remove printer from system settings menu
2. Go to prusa list

### Actual results

No Prusa Mk4s profiles available

### Expected results

Profiles being available since they were added in commit #8279 which is before the build's commit

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18724936/log.zip)

(No project active)

### Checklist of files to include

- [x] Log file
- [x] Project file

hgreenhow, 2025-02-13:
I had the same problem. I discovered that if I deleted my existing ~/.config/OrcaSlicer tree, then on re-running the appimage, the MK4S printer variants were available.

filloax, 2025-02-14:
Can confirm this workaround fixed (got suggested to do so on the Discord), but it's still a bug if you have to do that probably.

1activegeek, 2025-02-17:
So I'd like to chime in that I'm having an issue getting these new MK4S profiles including the IS and HF variants. I have an MK4S with HF 0.4 nozzle and would love to get these profiles. However it seems something is stuck and only pulls the MK4 profiles and filament profiles. Looking at the commit I can see the profiles are clearly labeled as MK4SIS and HF variants, so these are not the correct profiles. Would love any direction here to address/fix this. I have attempted the workaround above (blowing away the directory) but even on a fresh setup, it's producing the same result. 

I'd also like to highlight as an aside, that searching for Prusa in the search, only shows the XL. It would be nice if this could be fixed that typing Prusa in the search would show the other machines as well. 

[image]

1activegeek, 2025-02-17:
Sorry - update - I see that this commit isn't part of the main branch. For some reason I thought it was already. I'll have to grab nightly it looks like to get this update. Will patiently await the actual update push to Main.

rcloran, 2025-03-23:
The search only searches some fields. I named the printers "MK4S" and "MK4S HF" to match the MK4IS profile, which means it won't show up when searching for "Prusa".

This could probably be retitled more generically to something like "Printer selection search should search all machine fields" ... I've tried to do a quick search of issues but can't find anything like that yet.

I'm guessing the change would be made at https://github.com/SoftFever/OrcaSlicer/blob/main/resources/web/guide/24/24.js#L226 , but I'm not sure.

filloax, 2025-03-23:
> The search only searches some fields. I named the printers "MK4S" and "MK4S HF" to match the MK4IS profile, which means it won't show up when searching for "Prusa".
>
> This could probably be retitled [...]

Actually, it didn't show up even when searching "mk" without the workaround.

rcloran, 2025-03-23:
> Actually, it didn't show up even when searching "mk" without the workaround.

Ah, OK -- does it now in the released version? I don't think you ever updated here after mentioning that you noticed that the commit wasn't in the version you were using. Or do you still have to delete your user config dir?

filloax, 2025-03-23:
> Ah, OK -- does it now in the released version? I don't think you ever updated here after mentioning that you noticed that the commit wasn't in the version you were using. Or do you still have to [...]

I couldn't retry the issue as it seemed to require starting from an install without the preset, then upgrading to a version with it

rcloran, 2025-03-23:
> I don't think you ever updated here after mentioning that you noticed that the commit wasn't in the version you were using. Or do you still have to delete your user config dir?

Oh, different person 🙈

> I couldn't retry the issue as it seemed to require starting from an install without the preset, then upgrading to a version with it

Gotcha.

When I was building out the MK4S profiles I ended up deleting my user config dir repeatedly for various reasons. It does seem like the profile management database is finicky about ... something, and I never really figured out all of the ins and outs of it :( I've been unable to get meaningful logging out of OrcaSlicer in this area of the code, and ended up doing a lot of trial and error. So it wouldn't surprise me if there's still some small piece of data somewhere in the profiles that's somehow triggering the problem that you see.

rcloran, 2025-03-23:
#9050 makes the search work a bit better, but doesn't address the rest of this issue.