OrcaSlicer/OrcaSlicer #8365: A slight change to seam painting colours for the colourblind

FeetOffTheTable, 2025-02-10:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Hello, I'm red-green colour deficient (the most common type), and specifically with Support Painting I have a really hard time seeing the difference in colours between the 'detect overhang' red and the 'support painting' green. It doesn't help that overhangs occur underneath the model which naturally darkens the colour with an applied shadow effect, further muddying the colours. Attached is an example, and even with my high contrast QHD monitor I can barely tell the difference in colours.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add an option in the support painting interface menu to change the colours to something highly contrasting.

### Describe alternatives you've considered

1. Changing my monitor's colour palette with display accessibility options. While an option, it's not a very effective option for me personally and most likely others in my situation. Colour shifting the whole monitor can cause as many problems as it solves.
2. Colour correcting glasses e.g. Enchroma. Another option, albeit not effective on everyone and not practical. Also expensive.
3. Ask the wife for help. A very effective solution, but also highly susceptible to gaslighting.

igiannakas, 2025-02-10:
@yw4z maybe something to consider?

yw4z, 2025-02-10:
actually this could be nice touch but it requires a global setting on preferences. many green/red usages on UI like axis colors for X-Y. i can change values but i have no idea to how to add a setting to preferences window

igiannakas, 2025-02-10:
I can help with that :)

igiannakas, 2025-02-10:
This Pr here also added a preference in the settings window so something similar can be done for this too I think: https://github.com/SoftFever/OrcaSlicer/pull/8268

yw4z, 2025-02-10:
interesting part there is no app to manage color filters per app. Or any os manages it for per app. I assume its possible to activate filter only while specific app active with autohotkey. Since its a color filter that should be solve on that level. btw here is an AHK script to archive that

```
#Requires AutoHotkey v2+
#SingleInstance Force
#NoTrayIcon
Persistent(1)

DllCall("RegisterShellHookWindow","Ptr",A_ScriptHwnd)
OnMessage(DllCall("RegisterWindowMessage","Str","SHELLHOOK"),Watchdog)
Watchdog(wP,lP,msg,hWnd){
    try If(wP=1||wP=4||wP=32772){
		if(lP!=0){
			if(WinGetProcessName(lP)="orca-slicer.exe"){
				ActivateColorFilter(1)
			}else{
				ActivateColorFilter(0)
			}
		}
    }
}

ActivateColorFilter(value:=1){
	if(RegRead("HKCU\Software\Microsoft\ColorFiltering","Active") != value)
		SendInput("^#{c}")
}
```
this requires keyboard shortcut option enabled for color filter. I'm sure there should be an app that works per app

If i understand correctly there are different types of color blindness. every other cant see different color. so fixing only red not fixes all problem

btw color filter shortcut is win+ctrl+c. you can also asign it to mouse or f12 key for easy access

yw4z, 2025-02-10:
as an another option we can change color palette for
• overhangs
• Paint
• bottom of model

but i have no idea which colors will give better result for all types of color blindness

@FeetOffTheTable do have any more areas that makes hard to use?

yw4z, 2025-02-10:
I made a little research about situation. understood problem much more now. UI is hard to understand when you cant see tone of color if shade is same
proper colors or good contrast required to fix that

Color combinations to avoid
• green-red
• green-blue
• green-brown
• green-black
• green-grey
• blue-grey
• light green-yellow
• blue-purple

i assume we can ignore blue combinations because google tells its very rare. so we got
• green-red
• green-brown
• green-black (i guess we can ignore that too because it has contrast)
• green-grey (if they have same shade)
• light green-yellow (if they have same shade)

UI areas uses green-grey & green-brown
• Support painting
• Seam painting

UI areas uses green-red (Do you have any problems with this section)
• Axis colors
• Transform widgets / gumballs

Let me know if there is more places that can be solve
like you said its hard to solve sliced process legend since there are many colors in it

Happy to help if you point which parts of UI requires changes. i can tune colors slightly to fix this issue

I will send few examples as screenshots as examples to get a confirmation from you

Also i made some experiments with using grey-scale color filter of windows.
As you pointed i cant see any difference while painting. Green have same shade of grey also its not visible when mixed with red

igiannakas, 2025-02-10:
@yw4z youre awesome 👌🏻🎉

FeetOffTheTable, 2025-02-11:
@yw4z Ive just had a look through the app again to see what I can find. I have placed some screenshots of areas where improvements could be had, but just note that I'm not asking for an entire change of design just to accommodate the colour deficient, it would just be nice to have options. I think you've hit the nail on the head with the colour shade being the issue, as in colours that show up in similar grades when shifted to gray cause problems for the colour deficient and monochromatic. Red-green effects up to 8% of the male population so it's likely that there are many who would appreciate this help.

**The axis / transformation / scale widgets:**

When placed on the models they do not pose much of an issue to me as far as practicality. The widgets don't necessarily need to be identified through colours to be functional for me, and other applications do not always have them such as this example from Fusion:

[image]

However, the axis shown on the bottom-left corner of the plate is a bit muddied and much harder to tell at a glance:

[image]

So it would benefit me to stay away from that particular shade of Red / Green, if only to be able to tell X/Y apart at a glance. Not an immediate priority though.

**Seam painting:**

Not too much of an issue as it's being painted on to a gray base, but I can however see it being a problem for monochromatic people. I will also note that the "shadow" effect that occurs under the model is making it very hard to see where I have painted as the green is losing it's highlight effect from blending with the shadow, but this is probably an issue visually for everyone not just myself. 

[image]

If possible, just for quality of life improvements in general, a better highlight effect could be appreciated.

**Preview screen:**

As mentioned before, there is too much colour to account for in this screen, but the option to toggle 'display' makes it very easy to isolate areas that I need to focus on so I don't have an issue that can't be resolved over time and care.

I just want to note the only potential issue I have with this screen is that the "overhang" colour is very similar to the "bridge" colour, so if I'm not being careful sometimes I will incorrectly assume something is a bridge and cause a print failure. Again, this issue is not an absolute priority because it's something that results from my carelessness as well as lack of ability to distinguish blue from purple. 

[image]

I do not wish for the app to be recoloured, nor do I think that everything on this tab needs a colour overhaul because it's impossible to account for every possibility of colour combinations for all types of dichromatic people etc. But if it were possible to personalize the colour key, I could save myself a few failed prints.

I will note that I really appreciate that the other preview screens such as "Line Width" use an ascending gradient scale from blue to red, which makes it easy to visualize what's happening even for myself.

[image]

**Cut:**

Once again not the highest priority because CUT mode is usable even without colour indicators, but I will just note that "upper part" and "lower part" colours are very difficult for me to distinguish:

[image]

I will keep looking around the app for anything in particular to note.

yw4z, 2025-02-11:
Please don't hesitate send more if you found more areas

• Priority is on fixing Support / Seam painting
• Using red to blue instead full color spectrum is valuable information but i have to check results
• Axis colors, will investigate better colors but no promise. Checked what did big companies but found nothing. They use same raw rgb colors. I found some recommends using orange instead red and use a bit bluish tone of green. here is a chart

[image]

but using orange on x axis will confuse others. so will look another options

• Preview section is complicated since it uses many colors. but nice point about "overhang" and "bridge" bridge has very similar tones. They should use contrast colors since they positioned together. Also like you said its important to define them easier for better prints

• Using colors on cut section required since it creates a differentiation. Makes easier to understand which part is which one. especially when using rotated cut. Magenta-Cyan combination also used on measure gizmo. checked colors on gray scale and they uses very different shade. Do you really struggle on this combination? shifted colors a bit on image on below. Does that makes any improvement? 

[image]

FeetOffTheTable, 2025-02-11:
@yw4z you know, it's strange but now that I'm looking at this post on my phone the next day, I'm actually seeing very little issue with the original "cut" sample compared to before. The adjustments you made to the colours on that picture actually made it worse for me compared to the original. It could just be a difference in how phones display colours compared to my QHD IPS monitor, but regardless I'm going to triple check the colour balance on my monitor when I get home (another example of why messing with colour palette settings for accessibility reasons can cause other issues). All of the other images appear about the same to me so it just seems like I was having some trouble with that magenta colour before, but you can disregard my suggestion for changing the "cut" colours, I would only consider it a very minor inconvenience.
  
Chart looks nice, on palette 1 I can't easily discern 1 and 8 from each other easily. I feel like I want to explain part of what colour deficiency does to me because it's hard to explain why some things are more of an issue than other things until you experience it. Sometimes colours are so close that they can almost never be discerned as different, other times they are fairly close but you can 'train' yourself to see it properly. What I mean is that context can sway which part of the spectrum colour leans to in your mind,. E.g. STOP signs on a road will always appear red to me even if it's painted a colour similar to red, because that's what my brain is trained to know that colour as even if it doesn't appear that way at first. Things like Support Painting is that first issue, where it's so close I can almost never tell the colours apart even by staring at it. Other issues are the second type, where I can tell the colour if I'm really focusing or training myself to know what it is, but if I lose focus or only see it at a glance, I can misinterpret it. This is why I brought up the 'Cut' issue, because at the time I was not "seeing" the magenta properly, but now that I am it's a non issue. This is all to say that some of the issues above are more of a nitpick than others, which is why I've mentioned that it's not a priority. I think from now on I will try to grade the issues on a scale of difficulty to make it easier to understand, it's not always that I can't tell a colour apart in a vacuum. I hope that makes sense.

numericOverflow, 2025-02-13:
@FeetOffTheTable - YES, YES, YES.  As a severely color deficient dichromat, 1000% this!

To yes and their comments... personally I don't care WHAT the color is, it becomes difficult when colors are too similar for me to distinguish between them.  

Continuous gradients are the absolute worst. It's very difficult to match when there are no "gaps" in possible colors to help me differentiate.  

BTW - It was mentioned above, but should we open a separate ticket for the color schemes in the slice summary.  It's nearly impossible to get much information from that because of the continuous color gradients.

In this example, the line speed here is somewhere between 53 an 101 mm/s as far as my incomplete color vision can tell.  Throw dimming and 3D lighting aesthetic effects in there and there is a ZERO percent chance I can interpret the colors.
[image]

numericOverflow, 2025-02-13:
> Please don't hesitate send more if you found more areas
> [image]
> ...

@yw4z - THANK YOU! It's rare for color-sighted UI designers to be sympathetic to the plight of the color-deficient!

I'd echo that colors pallet_1-1 and pallet_1-8 are tough to discriminate.  The 2nd pallet is much better IMO.

FeetOffTheTable, 2025-02-13:
@numericOverflow I'm glad you've chimed in to provide your view, I was beginning to wonder if I was exaggerating the issue. On the thread I originally asked for help with this, I did have another red-green dichromat chime in that they agree with the struggles I had, and this is how I was directed to post here. 

I would be happy just to know I've somehow helped others like myself where normally it's not considered as much of a disability in terms of support provided. I appreciate knowing that it's becoming more of a thing that's support is being provided for, and that there are people like @yw4z who go out of their way to understand an issue before addressing it. I just hope I can somehow get programs to stay away from the palette-shifting wheels that say "deuteranopia" etc. but don't really address the issue.