OrcaSlicer/OrcaSlicer #8366: Flow rate changed to 2mm³/s on restart

crysxd, 2025-02-10:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

15.3

### Additional system information

M1 Pro MacBook Pro

### Printer

/

### How to reproduce

1. Create new project
2. Set Flow rate to 15mm³/s
3. Exit Orca
4. Reopen Orca
5. Create new project

https://github.com/user-attachments/assets/4791d188-933a-4968-82e2-644f87ccee49

### Actual results

Flow rate reset to 2mm³/s

### Expected results

Flow rate stays at 15mm³/s

### Project file & Debug log uploads

[log 2.zip](https://github.com/user-attachments/files/18730811/log.2.zip)

[Project.zip](https://github.com/user-attachments/files/18730815/Project.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

This only happens with this one filament at the moment, but I've also observed it with other ones. At some point it actually saves the value

Noisyfox, 2025-02-10:
Because you didn't save.

crysxd, 2025-02-10:
> Because you didn't save.

Didn't save the project?
Even if I save the project and then reopen the same project after a restart it's back to 2, in the project it then shows the 15 still of course but with the orange arrow to revert the "change"

[image]

Noisyfox, 2025-02-10:
> > Because you didn't save.
> 
> Didn't save the project? Even if I save the project and then reopen the same project after a restart it's back to 2, in the project it then shows the 15 still of course but with the orange arrow to revert the "change" back to 2
> 
> <img alt="Image" width="812" src="https://private-user-images.githubusercontent.com/5859228/411503693-90495051-13f7-4305-a9f6-d18aac356d54.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzkxODQxNTcsIm5iZiI6MTczOTE4Mzg1NywicGF0aCI6Ii81ODU5MjI4LzQxMTUwMzY5My05MDQ5NTA1MS0xM2Y3LTQzMDUtYTlmNi1kMThhYWMzNTZkNTQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDIxMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAyMTBUMTAzNzM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzhmYTAzNjI5ZTZjZjQwY2I1NTQ3NDRhOWJiMGNhZDA5YmRkNjQ0OTJmY2YwNmQ4YzdjY2VlMTIxMmIyYmU5NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.tgcgNd6sXWPQJx6GBJT2No1C8uB7lkKaqTQfwErjng4">

I don't see any issue here. That's exactly what it should be.

Noisyfox, 2025-02-10:
If you eant to save the config globally, you have to click the save button next to the profile name.

crysxd, 2025-02-10:
@Noisyfox I of course do that and it works for all other filaments. Please review the video I posted. The value is 15 without the orange arrow so the value is saved globally. 

I'm well aware how to setup and customize filaments in Orca, I use it for a long time and have dozens of filaments configured

1. The value is 15 and there is no orange arrow, so the value is saved
2. I also save a project
3. I close Orca and restart it
4. Open a new project or load the saved project
5. Now
 - The value is 2 in a new project
 - The value is 15 in the loaded project with a orange arrow, so it's just saved inside the project but not globally

crysxd, 2025-02-10:
Here, I save in the recording, restart and the value is reset. 

https://github.com/user-attachments/assets/b45c3aae-5e9e-469a-a73f-73ce49543de0

Noisyfox, 2025-02-10:
I see. Could you upload the entire orca config folder? It can be found in help menu. Thanks.

Noisyfox, 2025-02-10:
> The help menu item brings me to `/Users/christian/Library/Application Support` and selects `OrcaSlicer` but there is also a `Orca Slicer 2` folder. I zipped both but think the `OrcaSlicer 2` is unrelated.
> 
> [OrcaSlicer 2.zip](https://github.com/user-attachments/files/18732714/OrcaSlicer.2.zip) [OrcaSlicer.zip](https://github.com/user-attachments/files/18732713/OrcaSlicer.zip)

Cool thanks, I'm able to reproduce this issue. Will take a look soon.

crysxd, 2025-02-10:
Amazing, thank you @Noisyfox! This is haunting me for quite some time :D

Noisyfox, 2025-02-10:
Ok I found the issue, it's a limitation of current profile system, that a user-defined preset cannot inherit from another user-defined preset, which in your case, the `Nobufil ABSx` is inherted from `FormFutura TitanX ABS` which is another user preset.

Some details:
Although it looks like that the save is broken, but actually the broken part is the profile loading process.
When the profile is saved, it will check it's values and only save those differ to its parent profile. In your case, `FormFutura TitanX ABS` already have `filament_max_volumetric_speed` set to 15, so it won't duplicate that value in `Nobufil ABSx`'s file.
However when you restart Orca, when it load the `Nobufil ABSx` profile, it cannot properly find it's parent profile (which is the issue), and the values are fallback to default ones.

Noisyfox, 2025-02-10:
I don't know how you got those profiles though, because it shouldn't be possible to create new user profiles that inherits from another user profile in the first place IMO.

crysxd, 2025-02-10:
Ok, that explains the issue...I honestly don't know how I did that. When I create a new profile I usually choose a very similar one and edit it, so I assume I copy the issue forward.

Would explain why I only have this with ABS profiles :D 

I just opened the config folder in VS code and replaced all the `"inherits": "xyz"` for my custom filaments with `"inherits": ""` and the issue is indeed gone....

No clue how this ended up in this broken state :(
Sorry for wasting your time and thank you for looking into is so promptly! I sent a Ko-fi your way :)

closed 2025-02-10 by crysxd