OrcaSlicer/OrcaSlicer #8372: Use zink to implement better support for Wayland on Nvidia

klylabs, 2025-02-10:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

When running orca slicer under Wayland with an NVIDIA GPU,the home tab and preview pane usually fails to load, the most commonly suggested workaround involves setting environment variables like:
```
_EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json 
WEBKIT_DISABLE_DMABUF_RENDERER=1
```
this workaround forces software-based rendering rather than a proper hardware-accelerated solution.

I'm somewhat sure that the issue lies somewhere within nvidia's OpenGL implementation as this issue has also been known to occur with other CAD apps like FreeCAD, with users being required to set `QT_QPA_PLATFORM=xcb` which basically Qt apps to run via X11/XWayland

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Use Zink to run OpenGL on top of Vulkan bypassing the problematic NVIDIA OpenGL stack while retaining hardware acceleration.

to achieve this, you just need to set:
`__GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json MESA_LOADER_DRIVER_OVERRIDE=zink GALLIUM_DRIVER=zink`
in addition to `WEBKIT_DISABLE_DMABUF_RENDERER=1`

I'm still utterly unfamiliar with the whole structure of orca slicer, but ideally i'd like this to be set programmatically inside the application,  i temporarily edited the build script to add these to AppRun when it detects from nvidia-smi that the drivers is > 555 but i'm not sure if this is really an acceptable solution long term

### Describe alternatives you've considered

1. [Continuing software rendering]
2. [Switch to X11 when using orca slicer]

### Additional context

[Zink](https://docs.mesa3d.org/drivers/zink.html) documentation link under Mesa

klylabs, 2025-02-10:
Without Zink:

https://github.com/user-attachments/assets/ba750b01-2b12-4ed9-b0ac-e6776108cd5b

With Zink:

https://github.com/user-attachments/assets/c7d5a8ff-dfe7-4d1c-a32a-453ace67ee96

predy3dprint, 2025-02-11:
Thank you very much. That is the game changer. Works on Arch Linux perfectly.

yobson1, 2025-02-11:
works wonderfully on arch with KDE/KWin thanks

klylabs, 2025-02-26:
#8373 has been merged

closed 2025-02-26 by klylabs

orthadoxtesla, 2025-05-25:
alright i'm not quite sure how i can actually implement this on my own system. Using Arch linux and want to get my system to work but having no luck.

klylabs, 2025-08-25:
> Hi Guys, I tried but orca-slicer crash after open the window
> 
> Specs: orca-slicer (tested all versions: orca-slicer, orca-slicer-bin, orca-slicer-git, orca-slicer-nightly-bin, orca-slicer-unstable-bin) ArcoLinux (Arch) KDE PLASMA Wayland GPU Nvidia RTX 3060Ti (All packages updated to last versions today with "Pacman -Syuu")
> 
> `WEBKIT_DISABLE_DMABUF_RENDERER=1 __GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json MESA_LOADER_DRIVER_OVERRIDE=zink GALLIUM_DRIVER=zink orca-slicer `

Try updating to 580.76.05 and use the appimage version, the aur package drops this patch in its entirety

you can just download the latest .appimage and click it to run

if you're still seeing crashes, try turning off all your network connections and relaunch orca slicer, if it's working, it should be related to this issue #7210