OrcaSlicer/OrcaSlicer #8376: RTSP custom camera sources

peterdev22, 2025-02-11:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

The custom camera source feature only supports feeds that can be opened in a web browser (HTTP) but many home IP cameras use RTSP as an option to view the video stream directly.

Currently when viewing a RTSP or RTSPS stream a white box is shown.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I think it would be beneficial for RTSP and RTSPS streams to be natively supported for users that want to use an IP camera to have a better quality video stream of their printer.

Supporting both HTTP (as OrcaSlicer currently does) and RTSP covers the majority of video streaming options.

#### Implementation
Not sure how this would be implemented - maybe OrcaSlicer could just do the conversion from RSTP to WebRTC/HLS using FFmpeg but there may be a performance hit.

Ideally OrcaSlicer would natively support RTSP streams for custom camera sources.

### Describe alternatives you've considered

1. Converting a RTSP stream to WebRTC/HLS using software running on a Raspberry Pi.
2. Using a separate app such as VLC alongside OrcaSlicer to view the RTSP stream.
3. Use a mobile device and [VDO Ninja](https://github.com/steveseguin/vdo.ninja) as an alternative camera (works, but is not really a solution if you want to use an IP camera to view your printer).

thilojaeggi, 2025-02-11:
I can't even enter a URL/the window disappears immediately.

peterdev22, 2025-02-11:
Yep - i've had that issue as well, as mentioned in #8115 and #5410.

My current workaround:

1. Copy the stream URL onto your clipboard.
2. Open up the custom camera source window and right click on the text field.
3. Paste URL in using the right click menu instead of keyboard shortcuts.

jessedmatlock, 2025-03-05:
Same issues - both with white box instead of custom camera stream and the inability to type/edit the URL field, it just closes if you do anything other than paste from the Right Click menu.
Any word on this??

Abe-Oddysee, 2025-05-20:
That's funny bambulab uses rtsps

mpv rtsps://bblp:password@192.168.178.55:322/streaming/live/1