OrcaSlicer/OrcaSlicer #8379: Add ability to specify a particular filament or filaments does NOT purge to support or flush object.

Rescuer542, 2025-02-11:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

If using support material or PETG vs PLA for supports it is high risk to use a flush object. If the support or other material is flushed into a flush object it may not stick, failing the flush object and potentially the whole print. Same with supports. If it ends up in the support a support may fail. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

An option in filament settings to ONLY purge that material instead of using it in supports, infill, or flush objects.

Alternatively could add a setting where interface filament is specified to simply not do it for interface filaments.

### Describe alternatives you've considered

Only 2 current options are higher waste not flushing into Support, Infill or Flush objects, OR don't use a support interface and end up with rougher quality overhangs.

mjfsch, 2025-02-14:
Agreed, i would also extend this to not flushing to infill as well.
Flushing the interface material to infill can leave you with weak layers in the print.

madcrokodile, 2026-11-05:
Reproduces on 2.5.0 RC.

1-  'Flush into objects' infill' enabled
2 - dedicated interface layer extruder

One of a layers was filled with interface material.

I guess it is easy fix, just filter i-face extruder from flush