OrcaSlicer/OrcaSlicer #8383: Concentric top/bottom layers not continuous in 2.2.0 (but they are in 2.1.1)

Pykkles, 2025-02-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro 22H2

### Printer

Ender 3v2

### How to reproduce

Slice an object in 2.2.0 with _Top surface pattern = Concentric_ or _Bottom surface pattern = Concentric_

### Actual results

The top/bottom surface is printed as several disconnected loops in a non-continuous order.  

[image]

### Expected results

Extrusion is continuous, as in Orca Slicer 2.1.1

[image]

### Project file & Debug log uploads

[ConcentricTopBottomNotContinuous.zip](https://github.com/user-attachments/files/18757810/ConcentricTopBottomNotContinuous.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

I'm guessing that this might be caused by #6184 / #6432 as I can see some staggering on the new seams as printed.

This could be the same issue as @m-sundman reported on #6898 but I'm not 100% sure.

m-sundman, 2025-02-18:
Neither is continuous, but 2.1.1 is monotonous. What we really want is for it to actually be continuous (like described in #6898), which would automatically also make it monotonous.

slothy89, 2025-02-19:
I made a similar post in [#6670](https://github.com/SoftFever/OrcaSlicer/issues/6670) looking for the Monotonic pathing.

It died a death of no interest.

skl111, 2025-03-17:
This looks like its related to this as well

[](https://github.com/SoftFever/OrcaSlicer/issues/6510)

Which died due to no interest as well

MxBrnr, 2025-04-05:
Still an issue in 2.3.0-stable and 2.3.1-dev build as of today.

fieroscotty, 2025-04-05:
[orca2.1.1.zip](https://github.com/user-attachments/files/19617779/orca2.1.1.zip)

[orca2.3.0.zip](https://github.com/user-attachments/files/19617781/orca2.3.0.zip)

Here is a comparison of how it worked in 2.1.1 vs how its working now in 2.3.0. Something broke after 2.1.1.

skl111, 2025-04-05:
Unfortunately this is a work as designed issue and I don't think the devs plan to add the option to use it like it used to be anytime soon.

fieroscotty, 2025-04-05:
> Unfortunately this is a work as designed issue and I don't think the devs plan to add the option to use it like it used to be anytime soon.

That’s too bad. I’ve heard several people switching back to cura, Bambu studio and Creality print because of this. They all still do concentric layers correctly.

skl111, 2025-04-05:
Yep I have to use bambustudio for my concentric prints as well.

Here is a link to a response from one of the developers when I noticed this change during nightly builds. 

https://github.com/SoftFever/OrcaSlicer/issues/6510#issuecomment-2298125161

henrivdr, 2025-04-06:
@skl111  I think that's a different issue.
I'm pretty sure Orca devs won't design Orca to have a bug which was fixed in other slicers.

skl111, 2025-04-08:
> [@skl111](https://github.com/skl111) I think that's a different issue. I'm pretty sure Orca devs won't design Orca to have a bug which was fixed in other slicers.

You are probably right, and I am probably off after looking at the comments I missed. 
It does create an unpleasant pattern on the bottom whereas it used to create one you couldn't see.

I have attached a picture of the result of BS vs OS using concentric for reference in case it helps with this issue. Same filament flow rate, retraction and PA values for a P1 series.

[image]

MxBrnr, 2025-04-10:
Yeah, I find the aligned seam on the bottom pattern to be much more desirable than this offset one. A true concentric pattern wouldn't need to worry about the seam being a strength issue.

m-sundman, 2025-04-15:
> A true concentric pattern wouldn't need to worry about the seam being a strength issue.

Actually any truly concentric pattern will have a problem with the seam, but a spiral concentric pattern wouldn't: #6898

MxBrnr, 2025-06-13:
Bump. This really needs to be fixed one way or another. maybe with a new spiral/concentric pattern option that doesn't do the staggered seam. Another user brought up this issue on Discord again today: https://discord.com/channels/1137181739773603922/1383194815419781120

fieroscotty, 2025-07-30:
> Thank you very much for reporting the issue. It should now be fixed in the main branch.

YES!! IT WORKS!!! Thank you so much! Now I can return to Orca Slicer. :)