OrcaSlicer/OrcaSlicer #8386: Creating supports for empty layers for use in Stacking 3D prints

the-guyver, 2025-02-12:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Hi, I would like to request a feature request for OrcaSlicer to create supports on empty layers when stacking 3D prints.
When stacking 3D prints, multiple instances of the same 3D print are stacked on top of each other and then later separated from each other.
To create the stack, an empty layer (negative space) separates each individual element of the stack.
However, creating the stack on top of each other, even with the empty layers, still creates issues in trying to pry the stack elements apart.
If OrcaSlicer can create supports in the empty layers, then the stack elements would be much easier to remove from each other.
Right now, OrcaSlicer only creates supports for overhangs and bridges, but not empty layers between stack elements.

A use case scenario for stacking is stacking the honeycomb storage wall system (https://www.printables.com/model/152592-honeycomb-storage-wall/files).
With stacking, you can make use of vertical space, with the limited surface area of your 3D printer bed.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The solution I would like to see is the implementation of creation of supports in between empty layers for "floating" parts (which is in this case, are stack elements) of an object as an extra option for the «Supports» section.

klylabs, 2025-02-12:
Can you provide a project with the stacked honeycombed walls as described?

seems to work fine on my end

[image]

discip, 2025-02-12:
@the-guyver
Maybe this could help: https://github.com/SoftFever/OrcaSlicer/pull/8212#issuecomment-2621162473

the-guyver, 2025-02-15:
> Can you provide a project with the stacked honeycombed walls as described?
> 
> seems to work fine on my end
> 
> [image]

Hi,
From your image, I'm not really sure, but it appears you just used a raft.
Stacking involves stacking multiples instances of the same object, by adding an empty layer between the stack elements.
Here is a video that explains more about the process.
https://www.youtube.com/watch?v=xs2urfM0MRM
The issue with the method in the video, is that stacking involves stacking the object on top of the same object, which can be difficult to remove.
However, if OrcaSlicer can create supports between the stacks, this would be much easier to remove.
As far as I know, OrcaSlicer can only create rafts, and supports for bridges and overhangs--not stacking.

the-guyver, 2025-02-15:
> [@the-guyver](https://github.com/the-guyver) Maybe this could help: [#8212 (comment)](https://github.com/SoftFever/OrcaSlicer/pull/8212#issuecomment-2621162473)

Hi, thanks for the reply.
After going through the section, it appears it involves a multi-material printer, which I do not have.

klylabs, 2025-02-26:
> Hi, From your image, I'm not really sure, but it appears you just used a raft. Stacking involves stacking multiples instances of the same object, by adding an empty layer between the stack elements. [...]

sorry, i just used the default support, using snug would remove all of the clutter though

in my original image, you can still see some orange below which should be the color of the original part

here's the result using snug supports

[image]

the-guyver, 2025-03-02:
> sorry, i just used the default support, using snug would remove all of the clutter though
>
> in my original image, you can still see some orange below which should be the color of the original part [...]

Hi, can you explain to me how did you do this?

jbowers01, 2025-11-26:
To piggyback on this request, could there be a "duplicate stacked" right click function that could automatically duplicate and stack objects with the correct distance and supports for optimum printing, as well as a stacking calibration menu item?