OrcaSlicer/OrcaSlicer #8387: P1s's head met the table(

Vladimir225225, 2025-02-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

OrcaSlicer_Windows_V2.1.1_portable

### Operating System (OS)

Windows

### OS Version

Windows 10 22,04

### Additional system information

Intel i3. 

### Printer

P1S

### How to reproduce

Good afternoon! Today there was a malfunction in the slicer and the printer head scratched the table and damaged the plate. I am sending a 3mf file. This is the printing of a vase without a bottom on a substrate of 2 layers or more - after printing the substrate, the head sinks below the table and spoils the plate!  

### Actual results

This is a gross bug of Bambu Studio 1.10.1.50 and OrcaSlicer_Windows_V2.1.1 are independent of the geometry of the model!

### Expected results

Please fix this in new versions of the slicer!

### Project file & Debug log uploads

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Noisyfox, 2025-02-12:
Any project file?

Vladimir225225, 2025-02-12:
Vladimir Gorbunov
 
  
>Среда, 12 февраля 2025 г., 12:30 PM +03:00 от Noisyfox ***@***.***>:
> 
> 
>Есть ли какой-нибудь файл проекта?
>— 
>Ответьте на это письмо напрямую,  просмотрите его на GitHub или  отпишитесь . 
>Вы получаете это, потому что вы создали эту ветку. Идентификатор сообщения:  <SoftFever/OrcaSlicer/issues/8387/2653118907  @  github  .  com>

Noisyfox, 2025-02-12:
I think you cann't upload attachments to github via email.

Vladimir225225, 2025-02-12:
https://drive.google.com/file/d/1V-UruP01U_k1N2RVRabABhJuCtqVRTBS/view?usp=sharing
 
 
Vladimir Gorbunov
 
  
>Среда, 12 февраля 2025 г., 12:56 +03:00 от Noisyfox ***@***.***>:
> 
> 
>Я думаю, что загружать вложения на GitHub по электронной почте нельзя.
>— 
>Ответьте на это письмо напрямую,  просмотрите его на GitHub или  отпишитесь . 
>Вы получаете это, потому что вы создали эту ветку. Идентификатор сообщения:  <SoftFever/OrcaSlicer/issues/8387/2653194930  @  github  .  com>

Noisyfox, 2025-02-13:
Vase mode doesn't work well if filament/tool change is involved, even if it's only used for raft. There is no easy way of fixing this at the moment.

Noisyfox, 2025-02-14:
Try PR #8404

Vladimir225225, 2025-02-14:
Okay, thanks for the alert! I hope this prevents the printer from breaking down due to a slicer error. Will it be included in the next version?
 
 
Vladimir Gorbunov

PR #8404 (2025-03-30): Fix vase mode with filament change
This PR fixes issues of vase mode, when you have raft printed in a different filament.
The layer that color change happens will forced to be printed in normal mode.

Fix #8387
closed 2025-03-30 by SoftFever