OrcaSlicer/OrcaSlicer #8392: There is no Bambu PETG HF Profile

Flamingo89, 2025-02-12:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

There is no Bambu PETG HF Profile in the filaments options

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Can you add the new filaments that Bambu has recently released?

thelegendtubaguy, 2026-11-08:
This has been available since Orca 2.2.0 for Bambu printers, and 2.3.0 for non-Bambu printers.  This should be closed.

raistlin7447, 2026-11-08:
Thanks for the request. Bambu PETG HF is available in current releases. Bambu printers have had it since 2.2.0. Since 2.3.0 it's in the Orca filament library (#8287), so every other printer can select it too. Closing as completed.

If you're on 2.3.0 or later and still can't find it in the filament list for your printer, comment here with your Orca version and printer model and we'll reopen.

closed 2026-11-08 by raistlin7447