OrcaSlicer/OrcaSlicer #8402: Can't build on Arch Linux: cannot allocate memory

MoltenHydrogen, 2025-02-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

Commit b4a7721

### Operating System (OS)

Linux

### OS Version

Arch Linux 6.13.2-arch1-1

### Additional system information

CPU: Ryzen 5 5600x
RAM: 16GB
GPU: Radeon RX 6300

### Printer

none

### How to reproduce

1. Git clone the repo
2. Run DockerBuild.sh

### Actual results

Build fails on Step 9/12, `RUN ./BuildLinux.sh -dr`

### Expected results

Build completes successfully

### Project file & Debug log uploads

Above this snippet is just like thousands of lines related to building dependencies:
```
510.5 [5088/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWGltf/RWGltf_GltfLatePrimitiveArray.cxx.o
510.5 [5089/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWGltf/RWGltf_GltfMaterialMap.cxx.o
510.5 [5090/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWMesh/RWMesh_CoordinateSystemConverter.cxx.o
510.5 [5091/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWMesh/RWMesh.cxx.o
510.5 [5092/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWGltf/RWGltf_GltfJsonParser.cxx.o
510.5 [5093/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWMesh/RWMesh_TriangulationSource.cxx.o
510.5 [5094/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWMesh/RWMesh_TriangulationReader.cxx.o
510.5 [5095/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWGltf/RWGltf_CafReader.cxx.o
510.5 [5096/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWGltf/RWGltf_TriangulationReader.cxx.o
510.5 [5097/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWMesh/RWMesh_FaceIterator.cxx.o
510.5 [5098/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWMesh/RWMesh_MaterialMap.cxx.o
510.5 [5099/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWObj/RWObj.cxx.o
510.5 [5100/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWMesh/RWMesh_CafReader.cxx.o
510.5 [5101/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWObj/RWObj_ObjWriterContext.cxx.o
510.5 [5102/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWGltf/RWGltf_CafWriter.cxx.o
510.5 [5103/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWObj/RWObj_CafReader.cxx.o
510.5 [5104/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWObj/RWObj_ObjMaterialMap.cxx.o
510.5 [5105/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWObj/RWObj_MtlReader.cxx.o
510.5 [5106/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWObj/RWObj_CafWriter.cxx.o
510.5 [5107/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWObj/RWObj_TriangulationReader.cxx.o
510.5 [5108/5109] Building CXX object src/TKRWMesh/CMakeFiles/TKRWMesh.dir/__/RWObj/RWObj_Reader.cxx.o
510.5 [5109/5109] Linking CXX static library lib/occt/libTKRWMesh.a
510.5 ninja: build stopped: subcommand failed.
------

 3 warnings found (use docker --debug to expand):
 - LegacyKeyValueFormat: "LABEL key=value" should be used instead of legacy "LABEL key value" format (line 2)
 - WorkdirRelativePath: Relative workdir "OrcaSlicer" can have unexpected results if the base image changes (line 64)
 - LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 77)
Dockerfile:71
--------------------
  69 |     
  70 |     # Build dependencies in ./deps
  71 | >>> RUN ./BuildLinux.sh -dr
  72 |     
  73 |     # Build slic3r
--------------------
ERROR: failed to solve: ResourceExhausted: process "/bin/sh -c ./BuildLinux.sh -dr" did not complete successfully: cannot allocate memory
```

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I have checked htop while the script was running and never saw my computer go past 6 gigs of memory usage. Even it it was, the script seems to check if I dont have enough memory anyway.

Noisyfox, 2025-02-14:
Try increasing the swap? Linking takes a huge amount of memory sometimes.

MoltenHydrogen, 2025-02-15:
Tried increasing memory on docker desktop, run into a different problem. Use latest release build, same problem. Switch to building a flatpak instead, new problem. Use latest release build, same problem. I'm starting to think that my computer just does not want me to use orcaslicer.

Here's the error that shows up when running flatpak build on either version.
```
CMake Error at /usr/share/cmake-3.29/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR system filesystem thread
  log locale regex chrono atomic date_time iostreams program_options)
  (Required is at least version "1.66")
Call Stack (most recent call first):
  /usr/share/cmake-3.29/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.29/Modules/FindBoost.cmake:2393 (find_package_handle_standard_args)
  CMakeLists.txt:415 (find_package)
```

MoltenHydrogen, 2025-02-15:
https://github.com/anarsoul/io.github.softfever.OrcaSlicer Although I still can't get it to build myself, I found prebuilt flatpaks that seem to work for me in case anybody else finds this issue and just wants orcaslicer running.

c2h5oh, 2025-05-18:
It fails to build even with 32GB RAM available with just 16 threads available - recent changes to the build system have increased memory requirements greatly.

c2h5oh, 2025-05-19:
> Could you try removing these two lines and see if it helps? 

If I do I get this `Your distribution does not appear to be currently supported by these build scripts`, which is why I have added those 2 lines in my PR.

`DISTRIBUTION=arch` doesn't change anything in the build process, it just ensures that https://github.com/SoftFever/OrcaSlicer/blob/main/linux.d/arch is called before building starts.

That file does 2 things:
1. uses arch package manager to install build dependencies
2. uses arch package manager to check if GTK3 is installed

All of those exit and are done before cmake is even called. No memory use impact.

closed 2025-08-26 by github-actions[bot]