OrcaSlicer/OrcaSlicer #8406: Mobile support.

Iwo24pl, 2025-02-14:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Will you make mobile support?

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

IPA for ios or apk for android

### Describe alternatives you've considered

Web-based orca slicer

tlhintoq, 2025-02-15:
Hahahaha
Sorry but... Hahahaha
Basically makes no sense.
99% of the features would be near impossible to work on an iPhone.
Grabbing move handles now is hard, let alone on a screen 20x smaller than my monitor with hard to hit thumbs instead of a mouse.

And lets say you slice - then what?  How are you getting that on an SD card for the printer?

I get that a lot of this generation doesn't even own a computer - but you're going to need to step up now that you've gotten into this realm of printing etc.

haunter1982, 2025-02-15:
Well... There are online Slicers Out there...

https://grid.space/kiri/

haunter1982, 2025-02-15:
Yes and No. Orca slicer that could be used on mobile or tablet or webbased. Would be a dream.
The online Slicers are reduced in Terms of Features.
As with the newer printers you can send GCode Files via WiFi or Lan, it would be perfect if it would be possible to slice downloaded models on your mobile and send it to your printer...

tlhintoq, 2025-02-17:
I think you misunderstand laughter versus humility (humble).
I said nothing unhumble.  But the idea of a full blown iPhone version of Orca on a cell phone is a laughable idea.  But that's got nothing to do with whether or not I am humble.

Other people NOT the OP (including yourself) are directing the description from a mobile Orca slicer to a mobile companion app.
So... yeah even others (including yourself) recognize its not realistic.

Lighten up - enjoy a belly laugh when they come along because we all know the world is way too full of stress and pressure and everyone coming down on everyone else for the smallest littlest thing.  Nobody needs that.

haunter1982, 2025-02-17:
Why should it be unrealistic? Sure, GUI has to be completly different. And it is nothing that is done quickly. But as you can, already, use CAD on your Mobile, why not also be able to slice and print?

EricDimitri, 2025-02-17:
> I think you misunderstand laughter versus humility (humble). I said nothing unhumble. But the idea of a full blown iPhone version of Orca on a cell phone is a laughable idea. But that's got nothing [...]

Please explain me what is your contribution to the enhancement request? This is a development site, not a chitchat place to laugh on others. Try to contribute, not to lough. Thanks!! ;)

Iwo24pl, 2025-03-05:
> Hahahaha Sorry but... Hahahaha Basically makes no sense. 99% of the features would be near impossible to work on an iPhone. Grabbing move handles now is hard, let alone on a screen 20x smaller than [...]

Erm… you know that it already exists but for Creality right? They added creality print to their app, plus Prusa also made a mobile version, so why it wouldn’t be possible. I have a laptop but I don’t always have it with me and my printer is on remote location, so it would be perfect for me. And transferring files? Like literally every printer now supports WiFi. Impossible? Like today’s phones can run AAA games so a slicer would probably work.

Splarkszter, 2025-03-09:
> Hahahaha Sorry but... Hahahaha

Because of people like you, many loose interest in contributing to OpenSource projects.

iSteeb, 2026-10-09:
I’ve been working on an iOS client (iPad to follow, once the features are all implemented) that uses the untouched orca engine, and a fully native iOS UI (that hooks into the orca engine with a bridge).

I’d love for this to be a PR, and published by Orca themselves. I think this would be the only way to get it on the App store, without side loading.

**@ the maintainers, what do you need, to consider merging a PR, or otherwise adopting this kind of project as an official (or endorsed unofficial) client?** 

@tlhintoq - you can eat your words ;)

This is my thread on reddit for some more context and evidence of interest from the community: https://www.reddit.com/r/OrcaSlicer/s/WKkwaKFAN1

Felix14-v2, 2026-10-09:
iOS...

Iwo24pl, 2026-10-09:
@iSteeb Could you post your work on GitHub?

iSteeb, 2026-10-09:
@Iwo24pl Yes, I'll update here once the repo is pushed, which will happen the earliest of 

- I get 100% feature parity and tidy up the build process so that someone cloning the repo actually stands a chance at building the app
- OrcaSlicer weighs in on what they want to see (foundations at least) to endorse this project.  This, mostly so that there is a clear direction on what needs to be done to publish it (and what publishing might look like).  AGPL licensing makes publishing impossible without permission.

Not sure how to get their attention RE: point 2... 

@Felix14-v2 ??

Iwo24pl, 2026-10-09:
@iSteeb I can help you with it.

Some ideas from me:

- **Make the build process automated from Github CI**.
- **Make the app simple to use, it doesn't need full feature parity, since it is a mobile app**.
- **Add a simple process to upload the print to the printer**.
- **GUI for Moonraker, like Mobileraker - built into Orca Slicer**.

iSteeb, 2026-10-09:
I'm aiming for full feature parity (where logical... there's a handful of things that are really only applicable to desktop).

Aside from the build process (which is a whole separate beast I need to deal with, hopefully after Orca gives some indication on the chances of them allowing this to be published on the App store), the rest of your comment is done.  I've already printed from the phone (mostly default settings for my printer and filament choices).

GLM and Claude are iterating on the feature parity as my usage and attention allows, and in the next couple of weeks (if not earlier), they should be done.

Felix14-v2, 2026-10-09:
> [@Felix14-v2](https://github.com/Felix14-v2) ??

Targeting iOS seems weird to me. But anyway, here's another reddit post about porting Orca to mobile platform (Android this time): https://www.reddit.com/r/OrcaSlicer/s/cL2PqCI2ed

iSteeb, 2026-10-09:
iOS and iPadOS are, for all intents and purposes, the same. Why build for one and not the other?  UIs are scrollable and zoomable.

Thanks for the link - I’ll shoot them a message about this issue, maybe they join in the convo.

iSteeb, 2026-10-09:
Recorded this after today’s slice to show the latest demo/status (and conceptually prove a phone is not a bad interface to slice with, believe it or not!)

https://youtube.com/shorts/eCiiAPP2RSQ

@OrcaSlicer what would you need to see (technically) to consider endorsing (so that I can legally publish) or merging (and publishing yourselves) such a client?

I know I’ve jumped the gun a little considering I haven’t published my repo yet… will do so ASAP.

iSteeb, 2026-10-13:
https://github.com/iSteeb/OrcaPad repo is published.

https://testflight.apple.com/join/N8FnVMqF Testflight is under review.