OrcaSlicer/OrcaSlicer #8410: Thick bridges settings are inverted

henrivdr, 2025-02-14:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev nightly build

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04

### Printer

Creality CR-10SE

### How to reproduce

1. Add Cube primate.
2. Select any default profile.
3. Slice and view Line Type in the preview.
4. Zoom in to see the line width.

### Actual results

Bridge thickness is inverted from the settings.

### Expected results

Bridge thickness should match the selected settings.

### Project file & Debug log uploads

[Inverted_Thick_Bridges.zip](https://github.com/user-attachments/files/18803910/Inverted_Thick_Bridges.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

[image]

[image]

MxBrnr, 2025-02-14:
The `thick bridge` settings change the layer height, not the line width. What you are seeing is just a rendering representation of that.

henrivdr, 2025-02-15:
@MxBrnr The line width does change - inverted.  If this is intentional, I will close the Issue.

[image]

[image]
.
.
.
However , I guess the effective Flow is the key factor. And I see it does increase by 22%, even with the reduced width.

[image]

[image]

MxBrnr, 2025-02-15:
Must be some other setting influencing it because mine doesn't change.

[image]

[image]

closed 2025-05-25 by github-actions[bot]