OrcaSlicer/OrcaSlicer #8415: Orca Slicer under linux is painfully slow

officiallymarky, 2025-02-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Arch

### Additional system information

KDE / Wayland
Nvidia 3090 570.86.16
5950X 64G Ram Gen 4 NVME

### Printer

X1C

### How to reproduce

1. Open Orca
2. Open Model
3. do anything

### Actual results

Extreme lag when trying to move model on plate
Extremely lag when try to open context menu on objects to do things like change filament
Extreme lag when try to do things like modify scale %

Extreme lag = 1-10 seconds delay, making it extremely difficult to do things reliably.

### Expected results

It is not 100% of the time, some times it is manageable.  A lot of times it feels like I am trying to use top of the line software on an old Raspberry Pi.  Moving the model on the plate for example is painfully difficult as it is so laggy.  Just changing things like scale has delays to just type in the numbers (like 3-10 seconds).  

Usually I can manage to get what I need done, but I try not to do much in the slicer.  I am not 100% convinced it is limited to large models, but it may be something related to do that, I haven't really paid attention specifically to what model I am using when it happens, I try to just get it to the printer as fast as possible as it's just so slow to work with.

### Project file & Debug log uploads

I tried to upload but it keeps telling me failed to upload zip.
The model I was printing yesterday that was painfully slow was this one:

https://makerworld.com/en/models/413436?from=search#profileId-315446

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

It seems to be getting worse, when I first started using Orca on Linux, I only noticed it occasionally, now it seems to be fairly common.

vvuk, 2025-02-15:
I don't think this is a generic Linux issue, but something related to your configuration. Both 2.2.0 and latest nightly (85bbbbee) are fast here; Ubuntu 24.04, KDE, X11, AMD Ryzen 7 8845HS w/ Radeon 780M Graphics.

officiallymarky, 2025-02-15:
> I don't think this is a generic Linux issue, but something related to your configuration. Both 2.2.0 and latest nightly (85bbbbee) are fast here; Ubuntu 24.04, KDE, X11, AMD Ryzen 7 8845HS w/ Radeon 780M Graphics.

It isn’t limited to me, many on Reddit have expressed similar issues.

powermad80, 2025-02-15:
Known issue, you're on nvidia/wayland and I safely assume you're using the proprietary drivers. See #8145, there's people working on it.

officiallymarky, 2025-02-15:
> Known issue, you're on nvidia/wayland and I safely assume you're using the proprietary drivers. See [#8145](https://github.com/SoftFever/OrcaSlicer/issues/8145), there's people working on it.

I’m using open drivers

powermad80, 2025-02-15:
> I’m using open drivers

That may be the other answer for you. You're not encountering the usual nvidia/wayland incompatibility, but you're using drivers that have vastly diminished performance. Do you do anything else with high resource demand that works fine?

officiallymarky, 2025-02-15:
> > I’m using open drivers
> 
> That may be the other answer for you. You're not encountering the usual nvidia/wayland incompatibility, but you're using drivers that have vastly diminished performance. Do you do anything else with high resource demand that works fine?

The open drivers have been vastly improved and are the recommended ones for Linux these days since nvidia went open source.

klylabs, 2025-02-16:
Try to run this build

https://github.com/SoftFever/OrcaSlicer/actions/runs/13352791446,
specifically  OrcaSlicer_Linux_ubuntu_**2404**_PR-8373

from https://github.com/SoftFever/OrcaSlicer/pull/8373

and see if it improved your performance

officiallymarky, 2025-02-19:
> Try to run this build
> 
> https://github.com/SoftFever/OrcaSlicer/actions/runs/13352791446, specifically OrcaSlicer_Linux_ubuntu_**2404**_PR-8373
> 
> [image]
> 
> from [#8373](https://github.com/SoftFever/OrcaSlicer/pull/8373)
> 
> and see if it improved your performance

Any ENV I need to run when using this?

officiallymarky, 2025-03-08:
@klylabs zink solution drastically improves the performance for me.  It feels how it did on Windows.

closed 2025-03-08 by officiallymarky