OrcaSlicer/OrcaSlicer #8448: Objection Deletion

NateHackerman, 2025-02-18:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Sometimes you're printing a bed full of parts. Halfway thru the print, one part comes off the bed and turns to spaghetti, potentially ruining neighboring parts.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Creality Print has a great feature that allows you to stop printing individual objects. It's called "Object Deletion". From the device tab, you click on the object that you want to stop printing, click confirm, and the printer finishes the remaining parts, skipping everything associated with the deleted object.

### Describe alternatives you've considered

1. Use creality print just for the feature described above.

### Additional context

[image]

NateHackerman, 2025-02-18:
It seems this feature already exists within "Gcode Preview"? I could be wrong, the UI is not clear

MxBrnr, 2025-02-18:
This feature already exists. It is called "Exclude Object."

[image]

NateHackerman, 2025-02-18:
Yes. Closing as feature already exists.

closed 2025-02-18 by NateHackerman