OrcaSlicer/OrcaSlicer #8449: Painting straight lines

extremeodd, 2025-02-18:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

There is no current way to draw straight lines in Orca Slicer when trying to paint... well literally anything. Seam painting? No ability to draw a straight line from A to Z. Support painting? No ability to draw a straight line from A to Z. Color painting? Do I really need to type this out? 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Literally every single program that interacts with an image will give you some form of a straight line tool. This, for some reason, is absent from Orca Slicer despite using support/seam/color painting tools that are identical to something you'd find on any basic photo editing software suite. My request is to add another tool type to painting function that is a "straight line" tool. As in you click point "A" then click point "B" which will result in a straight line with a width that is determined by your pen size. This should be a fairly simple feature to code, the biggest issue is my inability to code.

### Describe alternatives you've considered

Have you tried using Mouse Keys? 

Yes, this does nothing to create a straight line. The line stops being created the moment you hit a key.

Have you tried using a random object to restrict your mouse's movement?

Yes, this does nothing for fluctuations in the movement of your hand (mostly roational).

Have you tried the triangle tool?

Yes, this does nothing yet again to select a straight line. Sometimes you get really lucky and a line will fall on a triangle, but most of the time it will not. Even on a flat surface.

Have you tried the Height Range, Fill, or Gap Fill tools?

Yes, yes and yes. None of them can draw a straight line from A to B.

### Additional context

Go into MSpaint, go to shapes and click on the "Line" button, click and hold the cursor anywhere on the canvas and drag to anywhere else on the canvas before releasing the click. This produces a line, I wish to do this in orca.

tlhintoq, 2025-02-19:
There is a way to do this.

Seam painting is keystroke `P`
Go in to seam painting.
Click and hold down the mouse button
Tap `P` to exist seam painting
Release mouse
Go to the end of your straight line
Tap 'P' to enter seam painting
Click the mouse (down and up)

You have a straight line.  Same with seam `L`

{Its an unexpected side effect of how mouse tracking is done in software.  As long as no developer goes out of their way to break it there's no reason it should go away.}

https://github.com/user-attachments/assets/8272d291-8e34-4026-9ccb-df5c237b8917

slothy89, 2025-02-19:
I mean, Seam painting has the "Vertical" option, which is part way there. And colour painting has both "Vertical" and "Horizontal" options.

But I suppose what you're after is the ability to have a straight line in any direction correct?

closed 2025-02-19 by extremeodd