OrcaSlicer/OrcaSlicer #8451: Can we please get the ability to add modifiers for fan speed settings?

dfbpurcell, 2025-02-19:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

This would allow us to create overhang fan speed calibration tests and dial in the quality of our overhangs

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add fan speed settings to the modifiers list, or better yet allow us to create custom fan speed profile presets so we can quickly apply them as modifiers to different objects

hugues974974, 2025-03-12:
Hi,
I really agree with this request.
I've tried all the week to set an ABS printing with certain parts presenting overhangs, and others verticals.
If I want to have a correct fan part ratio to avoid these overhangs, I need to push (in the filament settings) the part fan from 70 to 90% which is great for these overhangs but bad for the adhesion of other parts of the print (for ABS and ASA i'ts a shame !!!).
For now I'amnot at home but I'll come back soon with some printscreens and more explanations....

It's not (for now) possible to put a specific "modifier", acting on the filament settings (to set a different Fan% for the areas in overhangs...). It would be a real bid deal to get this !!
Best regrads,
Hugues (BBL X1C and a "personal" IDEX printer (from originaly BIQU B1 SE PLUS)

hugues974974, 2025-03-12:
Hi Again,
Just to precise what I've told before, here is a print screen showing the part which should need a certain amount of fan (70 to 80 %) to get a good aspect for the overhang (in the red circle), but not all the rest of part (It's ABS, so very sensible for adhesion layer...).

How can I do this without a specific adjustement of the part fan using a "modifier" at these layer  ?

Best regards,
Hugues
[image]