OrcaSlicer/OrcaSlicer #8455: Request: add K1 SE additional profiles from CrealityPrint

sven337, 2025-02-19:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Unless I am mistaken, the only profile available for K1 SE following  https://github.com/SoftFever/OrcaSlicer/pull/7002/ is "0.20mm Fast @Creality K1 SE 0.4"
There are more profiles available in CrealityPrint such as "0.16mm Standard@K1 SE", would it be possible to add it to OrcaSlicer ?

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Add the profiles from CrealityPrint? The files are attached.
I can help with that, but am not sure what you normally do in terms of importing profiles from other slicers.

[0.16mm Standard@K1 SE.json](https://github.com/user-attachments/files/18863954/0.16mm.Standard%40K1.SE.json)
[0.20mm  Standard@K1 SE.json](https://github.com/user-attachments/files/18863955/0.20mm.Standard%40K1.SE.json)
[0.24mm Standard@K1 SE.json](https://github.com/user-attachments/files/18863956/0.24mm.Standard%40K1.SE.json)

While we're at it, would it be worth adding profiles for the Creality materials?

[CR-ABS-1.75.json](https://github.com/user-attachments/files/18863980/CR-ABS-1.75.json)
[CR-PETG-1.75.json](https://github.com/user-attachments/files/18863991/CR-PETG-1.75.json)
[CR-PLA-1.75.json](https://github.com/user-attachments/files/18863987/CR-PLA-1.75.json)
[CR-Silk-1.75.json](https://github.com/user-attachments/files/18863983/CR-Silk-1.75.json)
[Generic ABS-1.75.json](https://github.com/user-attachments/files/18863981/Generic.ABS-1.75.json)
[Generic ASA-1.75.json](https://github.com/user-attachments/files/18863986/Generic.ASA-1.75.json)
[Generic PETG-1.75.json](https://github.com/user-attachments/files/18863988/Generic.PETG-1.75.json)
[Generic PLA-1.75.json](https://github.com/user-attachments/files/18863985/Generic.PLA-1.75.json)
[Generic PLA-CF-1.75.json](https://github.com/user-attachments/files/18863982/Generic.PLA-CF-1.75.json)
[Generic PLA-Silk-1.75.json](https://github.com/user-attachments/files/18863984/Generic.PLA-Silk-1.75.json)
[Generic TPU-1.75.json](https://github.com/user-attachments/files/18863992/Generic.TPU-1.75.json)
[HP-TPU-1.75.json](https://github.com/user-attachments/files/18863993/HP-TPU-1.75.json)
[Hyper ABS-1.75.json](https://github.com/user-attachments/files/18863994/Hyper.ABS-1.75.json)
[Hyper PETG-1.75.json](https://github.com/user-attachments/files/18863989/Hyper.PETG-1.75.json)
[Hyper PLA-1.75.json](https://github.com/user-attachments/files/18863995/Hyper.PLA-1.75.json)
[Hyper PLA-CF-1.75.json](https://github.com/user-attachments/files/18863990/Hyper.PLA-CF-1.75.json)

bistory, 2025-02-19:
> [@cochcoder](https://github.com/cochcoder) [@bistory](https://github.com/bistory) maybe you have suggestions here?

K1 SE is basically a K1C without panels and filtering system, I can make it by this friday 🙂

sven337, 2025-02-19:
Thanks! FYI the files I attached above were extracted from my CrealityPrint install in `~/.local/share/Creality/Creality Print/5.0/server_1/orca/default/parampack/K1 SE-0.4`.

bistory, 2025-02-20:
> Thanks! FYI the files I attached above were extracted from my CrealityPrint install in `~/.local/share/Creality/Creality Print/5.0/server_1/orca/default/parampack/K1 SE-0.4`.

Did you know that K1 SE is implemented in the current beta ? Install a nightly release and you'll have it :-)

sven337, 2025-02-20:
The only profile I can find in this git repository is for "0.20mm Fast @creality K1 SE 0.4", the others aren't there. Am I wrong?

commit 14f77e41 (2025-02-23): Add new K1 SE profiles (fixes #8455 (#8485)
closed 2025-02-23 by SoftFever