OrcaSlicer/OrcaSlicer #8457: Error "Coordinate outside allowed range" - only when painting supports.

tlhintoq, 2025-02-19:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev 2e4a2a2

### Operating System (OS)

Windows

### OS Version

Win10x64

### Additional system information

[image]

### Printer

Neptune 3 max

### How to reproduce

Job slices fine until I paint in supports.  
Then error as in title.

[image]

This job is no where near the size of the bed or height.  There's no reason even a skirt or tree expansion or anything like that would/should be a problem.

[image]

### Actual results

Error in title

[image]

### Expected results

Don't error

### Project file & Debug log uploads

[Shin.n3m.orca.zip](https://github.com/user-attachments/files/18865547/Shin.n3m.orca.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

Nothing else

tlhintoq, 2025-02-19:
Just found another factor:
When the disc and the part were no longer part of an `assembly` it sliced fine.
There's some magic combination of assembly and painted supports it doesn't like.

[image]

tlhintoq, 2025-02-24:
So now with 2.3.0 beta, its just crashing instead of throwing the error

vndk36, 2025-02-24:
Hello, I have encountered similar issue with the use of tree support in manual mode. If I change the tree support mode to be Hybrid, then it works but is the same as normal support basically. 

Here I have the project in 3mf format inside the zip file.
It should throw the same kind of error as @tlhintoq has.

[Bug project in beta orca.zip](https://github.com/user-attachments/files/18940113/Bug.project.in.beta.orca.zip)

vndk36, 2025-02-26:
Spotted the issue again with another model with tree support.

Only way to fix it, download the portable version of 2.2.0 and slice it there.

[wing MJF core v2.5 v2 project.zip](https://github.com/user-attachments/files/18985779/wing.MJF.core.v2.5.v2.project.zip)

vndk36, 2025-03-03:
Coming back to this, for @tlhintoq and me, it looks like the last beta (beta 2 of version 2.3.0) fixes the wholes issue.

closed 2025-06-09 by github-actions[bot]