OrcaSlicer/OrcaSlicer #8459: Request for a native ARM64 build (improved performance on Surface devices)

Basti77, 2025-02-19:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Hello,

I’d love to see an official ARM64 build of this application. I recently bought a new Surface device running Windows on ARM, and using the x86 emulation often impacts both performance and battery life. Given that this application is primarily used in outdoor settings, having a more power‐efficient, native ARM64 version would be ideal for extending battery life and overall usability.

Thank you for considering this request!

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

I and most people don't know how to compile such a project themselves.

Fritomaster, 2025-03-28:
Yes please!

boswelja, 2025-06-22:
I tried building on my Yoga Slim 7x, but it seems like some dependencies used today aren't ready to be compiled form ARM - I've had some success bumping some of them, but that'd need a fair bit of testing I'd imagine.I'll see about opening some PRs for bumping some deps to get us inching closer to this!

Disclaimer: I have zero prior experience with cmake, don't get your hopes up too much 😅