OrcaSlicer/OrcaSlicer #8474: Separate prime towers for multi material prints

the-robo, 2025-02-21:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

On multi-material prints, I'm always frustrated when prime towers break due to dissimilar filaments. Often this will ruin the print. For example: Printing in PLA and using PETG for support interfaces. This works on the model, but sometimes the prime tower will break because PETG does not adhere well to PLA. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

By printing separate prime towers per filament type, they are less likely to break mid-print and cause the print to fail.

As well, separate prime towers would help with filament recycling, as the filament would not be mixed.

### Describe alternatives you've considered

There are no alternatives currently.

Mike-Flexicon, 2025-02-21:
This feature is literally simplify3d's only redeeming quality. If they had any kind of decent fan control for overhangs and the like, I'd keep using it. But right now I just agonize over the fact that prusaslicer/orcaslicer/simplify3d each has something I need/want to make this idex printer work well.

simplify's purge tower, orca's 3 wall print order, and prusa's mat'l/print settings control.  Combine the best of each and you'd have the one to rule them all.

/rant

juanemartinboix, 2025-02-23:
At least add a way to configure the number of lines that get printed from each filament so the tower does not fail when using detachable support material (petg+pla for example). 
Being so multiextruder related ,I´m thinking on trying to make the tower(s) as a filament start gcode or a Klipper Macro....until the prime tower in Slicer evolves.

wwmotorsport, 2025-04-16:
I agree, something like this is certainly needed for mutli-material prints where one is used for support (PLA and PETG for example). We are running an IDEX Modix Big180X and our purge towers constantly break somewhere during longer prints since the PLA and PETG just do not stick together when printed on top of one another.

Perfect world, there would be separate purge towers for each material type, but even if the purge tower stabilization cone for the tower just had concentric patterns so that each material was adhering to itself from one layer to the next that would be better. 

We have found a workaround for this as outlined below:
1) Manually add one cylinder per material type that are the same height as the last material change
2) Apply one material to each cylinder
3) Make sure that the added parts are before the printed parts in the object list
4) Select "Intra-layer order" to "As object list" 

Taking all of those extra steps forces the basic behavior we are looking for, and the cylinders are effectively prime towers due to the forced printing order established in step 4. However, that is a lot of extra manual steps to take, so it would be great if some of that functionality could be added into the pre-built prime tower options.

juanemartinboix, 2025-04-16:
Beautifully explained !!
Thanks for the workaround, now I know how to do it.

wwmotorsport, 2025-04-17:
For what it's worth, it does not seem like this works well if you have "Independent support layer height" or "Tree support adaptive layer height" enabled.

hrisand, 2026-04-25:
Hi All,

I am struggling also with the wipe tower stability, you can use specific tool\filament for the outer layer and make it more stable but sometimes the internal layer of purge does not stick well and hits the nozzle, depending how bad it gets may just spread debris on the plate, knock the wipe tower or cause a layer shift. I get it that requires more swaps, space and filament waste, but  when you have failed prints that happens also anyway.
The plate temperature should be the higher temp material value and the nozzle is changing anyway with the corresponding filament. The ability to choose the support material and adhesion compatibility matrix will be great. Like when you have two materials which do not stick to have by default Z distance of the support to zero and speeds adjusted. 

I am not sure when that will happen but now with more and more multitool changers and especially taller prints will become more common problem.

midicdj1000, 2026-04-25:
Hi, I wanted to make a comment about my recent experience with PLA - PETG supports. I did some testing and I found the type - brand of filament is just as important as the slicer settings. Mixing brands was giving me a lot of issues. Eryone PLA would stick to Elegoo PETG, but the Elegoo PETG refused to stick to the Eryone PLA no matter what I tried and I went through a lot of testing with speeds and nothing would work. I then tried a few other PETG and eventually had great success with Elegoo Rapid PETG and Elegoo PLA. It worked perfectly at most speeds with PETG on PLA and PLA on PETG. It was strong enough to print Iron man helmet parts over 14 hours 100% sitting on a PETG support. 

[image]

Green was Elegoo PETG + and red Eryone galaxy PLA the PETG wouldn't stick to the red PLA

I then changed to Elegoo rapid PETG (grey) and Sparkle RED

[image]

also Gold Sparkle

[image]

It worked perfectly with 3 14 hours prints all sitting on PETG.