OrcaSlicer/OrcaSlicer #8489: Website gives malware risk warning

LilBub, 2025-02-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

NA

### Operating System (OS)

Windows

### OS Version

Windows PC, Google Chrome browser

### Additional system information

Website gives warning 

### Printer

NA

### How to reproduce

https://orca-slicer.com/

### Actual results

[image]

### Expected results

No warnings!

### Project file & Debug log uploads

Screen shot attached

### Checklist of files to include

- [ ] Log file
- [ ] Project file

purp0w, 2025-02-21:
It says this in the [readme](https://github.com/SoftFever/OrcaSlicer/blob/main/README.md):
"🚨🚨🚨Important Security Alert🚨🚨🚨
Please be aware that "orcaslicer.net", "orcaslicer.co" or "orca-slicer.com" are NOT an official website for OrcaSlicer and may be potentially malicious. These sites appear to use AI-generated content, lacking genuine context and seems to exist solely to profit from advertisements. Worse, it may redirect download links to harmful sources. For your safety, avoid downloading OrcaSlicer from this site as the links may be compromised."

johnny-bit, 2025-02-21:
orca-slicer.com is NOT an official website and it's correcly labelled as potential malware.

An official website is in the works here: https://github.com/OrcaSlicer/orca-website

MattiasC, 2025-02-22:
Added to uBlock Origin yesterday:
https://github.com/uBlockOrigin/uAssets/issues/27270#issuecomment-2674168461

closed 2025-06-01 by github-actions[bot]

Sawtaytoes, 2025-06-09:
https://orcaslicer.org is also throwing a security warning for me.