OrcaSlicer/OrcaSlicer #8495: Draft shields are not generated when printing multiple objects by object

jpmgrove, 2025-02-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Lenovo ThinkPad T16, pretty vanilla

### Printer

Modified Ender 3 running Klipper

### How to reproduce

Add multiple objects to a plate. Enable Draft Shield. Set Print Sequence to By Object. Skirts and Draft Shields are not generated.

### Actual results

Draft shields and/or skirts are not generated

### Expected results

Draft shields and/or skirts would be generated, as they are when printing multiple objects by layer, or a single object.

### Project file & Debug log uploads

[Ender NG Parts.zip](https://github.com/user-attachments/files/18917989/Ender.NG.Parts.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

jpmgrove, 2025-02-22:
This only occurs if the brim type is not set to Auto. When setting to Auto, the draft shield is generated.

igiannakas, 2025-02-25:
Works in 2.3 beta:

[image]

Please make sure you select by object draft shield as differently you'll have a part collision with the draft shield when the second object is printed.

closed 2025-06-04 by github-actions[bot]