OrcaSlicer/OrcaSlicer #8504: CORE One Print Profiles

bwees, 2025-02-22:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Is it possible to get the CORE One Print Profiles imported into OrcaSlicer? I saw that @rcloran was able to import the MK4S with his scripts. I am having a hard time getting them to work for me otherwise I would do the PR myself. 

Thanks!

### Which printers will be beneficial to this feature?

Others, Marlin

### Describe the solution you'd like

CORE One Print Profiles added to orcaslicer

rcloran, 2025-02-22:
Those scripts are pretty rough. Here's how I ran them:

```
perl superslicer_to_orca.pl --input ~/src/PrusaSlicer/resources/profiles/PrusaResearch.ini --outdir
uneditedout --nozzle-size 0.999
cd uneditedout/user/default
mkdir -p out/{machine,process,filament}
cp ~/src/OrcaSlicer/resources/profiles/Prusa/machine/fdm_machine_common.json out/machine/
cp ~/src/OrcaSlicer/resources/profiles/Prusa/filament/fdm_filament* out/filament/
cp ~/src/OrcaSlicer/resources/profiles/Prusa/process/fdm_process_common.json out/process 
../../../process.py
```

I've found at least one thing missing since the original PR, #8486 

I'll happily accept pull requests to update the scripts for any missing options, make them easier to run, etc.

bwees, 2025-02-22:
Cool yeah I realized that I hadn't setup all of the proper export directories on my output directoy. Looks like the CORE One uses a different dependency system than the MK4S:

[image]

Will likely take quite a bit of work to get this converted unfortunately.

rcloran, 2025-02-23:
The way the scripts work (specifically the Python "post-processing" that I added) is that they go through each final file you want, rewrite it so that there's no inheritance at all ("concretize"), and then build up the inheritance according to any structure you want. 

Your new profiles don't need to follow the inheritance structure in Prusa's original profiles. In fact, they can't, because Prusa uses a pretty complex structure of multiple inheritance to make it all work, and Orca doesn't support multi-inheritance AFAICT. 

What I'm saying is... I don't /think/ that the inheritance structure that's there should cause you any additional time, but it will still take time to get it all right, of course.

bwees, 2025-02-23:
Yeah I took some time yesterday to see how it all works. It looks like Prusa just use the mk4s as the base profile and tweak the acceleration and speeds for each profile. Took about an hour to build the profile manually based on the mk4s. 

I had to do a few adjustments to the start gcode to handle chamber temperatures (orca only has one temperature setting instead of 2 like PS). I essentially made it so if the chamber temperature is set below 35c, then it uses the temp as a nominal temperature and if it's above 35c then it's the minimum temperature. Once I have my CORE one I will post the profile here once I am able to test. Kinda sucks that OrcaSlicer doesn't support bgcode but I should be able to add it as a post processing script.

Thanks for your help!

rcloran, 2025-02-23:
I tried to use Prusa's command line to do a post-process to transform into bgcode, but it failed saying the input gcode was invalid. I didn't try to figure out why. If you work out a solution, I'd be super interested to learn what you did to make it work :)

bwees, 2025-02-23:
Yeah I am hitting that same issue 😅. I will report back if I can get it working!

rcloran, 2025-02-23:
> I had to do a few adjustments to the start gcode to handle chamber temperatures (orca only has one temperature setting instead of 2 like PS).

If you've made these changes in code, it'd be great to add some sort of switch to the python scripts so that they can be re-run in the future for either printer. Happy to try keep the scripts all in one place by having that. That said, keeping your own repo with core one specific code so that anyone who wants to re-import in the future can use that is also a reasonable option.

bwees, 2025-02-23:
> If you've made these changes in code, it'd be great to add some sort of switch to the python scripts so that they can be re-run in the future for either printer. Happy to try keep the scripts all in [...]

Unfortunately I didn't use the output from the conversion scripts. Doing a bit of digging you can see that almost all CORE One Profiles inherit the MK4S Profiles. I was able to just make a new printer in Orca that was based on the MK4S and tweak 10 or so changes for the machine profile and about 10-15 settings per process profile. I just used the profile diff tool in PrusaSlicer to see the differences between the MK4S and CORE One profile. The profiles are pretty much the exact same between MK4S and CORE One except for acceleration and speed changes so it wasn't that bad to do manually.

bwees, 2025-02-23:
If anyone wants to try it, here is my WIP profile. THIS IS UNTESTED. I am not responsible if you break your printer. Please be cautious when printing with it. Let me know if any changes need to be made.

[Prusa CORE One 0.4 nozzle.zip](https://github.com/user-attachments/files/18932721/Prusa.CORE.One.0.4.nozzle.zip)

(might need to change extension to `.orca_printer`

bwees, 2025-02-23:
> I tried to use Prusa's command line tool to do a post-process to transform into bgcode, but it failed saying the input gcode was invalid. I didn't try to figure out why. If you work out a solution, I'd be super interested to learn what you did to make it work :)

Managed to figure out the issue. Their command line tool looks for specific strings in the gcode to mark the start/end of settings as well as the source of the Gcode. OrcaSlicer uses different strings and thus fails when trying to run the converter. I made modifications and it is able to produce a gcode file however PrusaSlicer throws an error when trying to preview the bgcode file. I'm not sure if the printers will throw an error since it seems to be related to slicer settings, not the actual print code. 

I put my changes up on https://github.com/bwees/orca_bgcode if you want to try it out. I don't have a bgcode supported printer yet so I can't test.

bwees, 2025-02-23:
> >
> I put my changes up on https://github.com/bwees/orca_bgcode if you want to try it out. I don't have a bgcode supported printer yet so I can't test.

I modified the script to work as a post-processing script (in-place modification and new filename hint). Instructions are on the README

bwees, 2025-02-25:
@rcloran managed to get access to a Prusa Mini. Was able to the bgcode conversion post processing script:

```/Users/bwees/Documents/OrcaSlicer/orca_bgcode/dist/bin/bgcode --post-processor --slicer_metadata_compression=1 --gcode_compression=3 --gcode_encoding=2;```

and the printer had no issues printing from the file. Should be good to use with other prusa printers that support bgcode.

P-C-R, 2025-02-26:
> Managed to figure out the issue. Their command line tool looks for specific strings in the gcode to mark the start/end of settings as well as the source of the Gcode. OrcaSlicer uses different [...]

Thats what i searched. Still sad that bgcode is not directly implemnted in orca

bwees, 2025-02-26:
> I have some Core One profiles aswell. should we compare?

I posted my profile [here](https://github.com/SoftFever/OrcaSlicer/issues/8504#issuecomment-2677066311), its missing the bgcode PP script but all of the parameters are done.

bwees, 2025-02-26:
> Thats what i searched. Still sad that bgcode is not directly implemnted in orca

It makes sense, Bambu Slicer was designed to just transport everything in 3mf files. Prusa is also the only printer that uses bgcode (without modification). There needs to be more industry adoption before it makes sense to invest the dev time

NateHackerman, 2025-03-18:
I have a Core One and would love to see this implemented. How can I support testing / implementation?

bwees, 2025-03-18:
> I have a Core One and would love to see this implemented. How can I support testing / implementation?

We are doing most of development on the Orcaslicer discord (channel: https://discord.com/channels/1137181739773603922/1341526631465881762). If you want to help out join us there and we can get you the latest config to try!

bwees, 2025-04-01:
We have moved development of the profile to a GitHub Repository: https://github.com/bwees/orca_coreone. The latest profiles are available in the releases section.

rinkek, 2025-04-15:
> We have moved development of the profile to a GitHub Repository: https://github.com/bwees/orca_coreone. The latest profiles are available in the releases section.

Thanks!
I downloaded v0.1.1 and it's not complaining anymore. I don't use the compiled code since i'm using ethernet, which is fast enough. 
Much appreciated!

bwees, 2025-05-04:
Merged with #9352. Closing.

closed 2025-05-04 by bwees