OrcaSlicer/OrcaSlicer #8505: Calibration -> Flow rate -> Normal & YOLO broken with 0.6mm nozzle and too low filament Max volumetric speed

mattaw, 2025-02-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

CPU: 5800x3d, 32 GB RAM, Nvidia 3090

### Printer

Custom Klipper - modified Tronxy XY 2 Pro w. Dragon High Flow hotend 

### How to reproduce

1. Edit filament "Max volumetric speed" to 2mm^3/s
2. Set 0.6mm nozzle
3. New project
4. Calibration -> Flow rate -> Pass 1 / Pass 2 / YOLO / YOLO precise
5. Slice plate
6. Inspect flow rate via color scheme image, poor quality flow test pieces produced as capped flow-rate prevents proper extrusion for automatically set 0.3mm layer height

For correct operation:

1. Edit filament "Max volumetric speed" to 2mm^3/s
2. Set 0.6mm nozzle
3. New project
4. Calibration -> Flow rate -> Pass 1 / Pass 2 / YOLO / YOLO precise
5. Slice plate
6. Inspect flow rate via color scheme image, correct flow test pieces produced for 0.3mm layer height

### Actual results

The flow rate calibrations pieces created are faulty due to max flow limited by the filament max volumetric speed. Most test pieces hit the limit early for 0.3mm layers (auto set from 0.6mm nozzle) and will be identically  underextruded and very rough. This affected me with a 0.6mm nozzle on a Dragon High Flow hotend with titan extruder and PETG.

In my opinion the best solution to this complex issue is likely a pop-up message in the slicer, or better wiki documentation mentioning this could be a problem with the tests.

### Expected results

Either a warning or better documentation to be able to diagnose this issue properly.

### Project file & Debug log uploads

[debug_Fri_Feb_21_16_18_56_23772.log.0.zip](https://github.com/user-attachments/files/18924203/debug_Fri_Feb_21_16_18_56_23772.log.0.zip)

[Orca YOLO Flow Calibration.zip](https://github.com/user-attachments/files/18924204/Orca.YOLO.Flow.Calibration.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

Thanks for the incredible piece of software! Its amazingly good.

theboee, 2025-02-23:
Already known bug, was referenced in #6989. Is on the roadmap and is slated to be fixed in a future release.

mattaw, 2025-02-23:
I appreciate the link to 6989! I actually found it an hour ago and linked to this issue, but maybe I should have linked this issue to 6989 like you did? I'm still working on my github skills.

closed 2025-02-23 by mattaw