OrcaSlicer/OrcaSlicer #8507: Segmentation Fault when editing motion ability

bwees, 2025-02-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta

### Operating System (OS)

macOS

### OS Version

macOS 15.3.1

### Printer

Any Printer, Video shows a Core One Profile I'm trying to build

### How to reproduce

M2 Pro, 32GB RAM

Editing motion ability parameters on any printer will cause a segmentation fault after changing value and hitting "enter" or "tab".

https://github.com/user-attachments/assets/98581043-523b-43cf-8aa7-1335cb2d84bb

### Actual results

Program Crashes

### Expected results

It doesn't crash :)

### Project file & Debug log uploads

Log:

[debug_Sat_Feb_22_11_31_19_10616.log.0.zip](https://github.com/user-attachments/files/18924441/debug_Sat_Feb_22_11_31_19_10616.log.0.zip)

Stdout:
```
[2025-02-22 11:25:45.956776] [0x00000001f1438840] [trace]   Initializing StaticPrintConfigs
loading network module at /Users/bwees/Library/Application Support/OrcaSlicer/plugins/libbambu_networking.dylib
after dlopen, network_module is 0x766dcb70
2025-02-22 11:25:47.056 OrcaSlicer[10392:3498605] +[IMKClient subclass]: chose IMKClient_Modern
2025-02-22 11:25:47.162 OrcaSlicer[10392:3498605] +[IMKInputSession subclass]: chose IMKInputSession_Modern
1. lo0   : 127.0.0.1
2. en0   : 10.245.150.246
3. utun7 : 100.85.233.125
4. en7   : 192.168.10.196

BambuTunnel::GetMsg(1)
UNSUPPORTED (log once): POSSIBLE ISSUE: unit 0 GLD_TEXTURE_INDEX_2D is unloadable and bound to sampler type (Float) - using zero texture because texture unloadable
[1]    10392 segmentation fault  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
```

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

See attached video. Sometimes it does save correctly but is extremely inconsistent.

bwees, 2025-02-22:
This bug is present in v2.2.0 as well

SoftFever, 2025-02-23:
It's very likely caused by the silent mode as I can see you have two columns there.
I can't guarantee the silent mode works properly in Orca. It's not actively maintained.
I also failed to get why there's a need to have two modes, it makes the logic very crumby IMHO. 

Can you try disable silent in your custom profile and give it a try?

bwees, 2025-02-23:
This profile was built off of the mk4s profile. There is no check box to disable/enable silent mode in OrcaSlicer so not an easy way for me to test this.

I don't think its because of silent mode as if I try changing the motion ability settings on the A1 or P1S profile, the crash still occurs.

SoftFever, 2025-02-23:
Let me have a look

SoftFever, 2025-02-23:
Should be fixed now.

commit 99e4879f (2025-02-23): Disable silent_mode as we don't support it. Fixes #8507
closed 2025-02-23 by SoftFever