OrcaSlicer/OrcaSlicer #8509: Generic PLA/Generic PETG mapping to AliZ filaments

bwees, 2025-02-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta

### Operating System (OS)

macOS

### OS Version

macOS 15.3.1

### Printer

Bambu P1S and A1

### How to reproduce

1. Set a filament slot on your Bambu printer **from the app (see below)** to Generic PLA or PETG
2. Do an AMS sync in OrcaSlicer

### Actual results

See that filament is loaded as AliZ PLA/PETG instead of Generic PLA/PETG

Also note, Generic PLA/PETG is missing from the filament selector in Device Tab:
[image]

### Expected results

Loads as Generic instead of AliZ, also should be able to select Generic PLA/PETG from the Device tab.

### Project file & Debug log uploads

[debug_Sat_Feb_22_13_09_08_12933.log.0.zip](https://github.com/user-attachments/files/18924950/debug_Sat_Feb_22_13_09_08_12933.log.0.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

zoroseerus, 2025-02-22:
same on my end

bwees, 2025-02-22:
Appears to be related to https://github.com/SoftFever/OrcaSlicer/pull/8496

JeffRudzinski, 2025-02-23:
Same thing for me on Bambu A1.

Sonicboom247, 2025-02-23:
Same here, i unchecked Aliz PLA and I can now see Generic PLA again.

kendygi, 2025-02-23:
Same on my Bambu Lab A1

NordicRavn, 2025-02-23:
Same here. Aliz pla for every pla profile and cant find general PLA

bwees, 2025-02-23:
Issue was fixed with #8531. Update to the beta-2 [nightly build](https://github.com/SoftFever/OrcaSlicer/releases/tag/nightly-builds), it has the fix implemented.

closed 2025-02-23 by bwees