OrcaSlicer/OrcaSlicer #8518: Unable to select dark mode in flatpak build

dannyglover, 2025-02-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Beta

### Operating System (OS)

Linux

### OS Version

Endeavour OS

### Additional system information

KDE

### Printer

Bambu A1

### How to reproduce

* Install Flatpak
* Launch Orcaslicer
* See that there is no option to enable dark mode and the app is stuck in light mode
* Try editing the config file to turn dark mode on, but orca overwrites the file on startup, reverting it.

### Actual results

Orca simply will not go into dark mode with the flatpak version. This doesn't affect the Appimage version however.

### Expected results

Orca should be able to put into dark mode when using the flatpak build.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

darco415, 2025-02-23:
I prefer non dark mode as now I can see my custom build plate! Anyways a feature to turn on/off dark mode like in past would be nice

dannyglover, 2025-02-24:
> I prefer non dark mode as now I can see my custom build plate! Anyways a feature to turn on/off dark mode like in past would be nice

Just to be clear, the dark mode checkbox exists on the AppImage build, and you can toggle dark mode on/off. 

It's just missing the toggle on the flatpak version.

dannyglover, 2025-02-27:
> I havent seen the toggle in OS in a long time? I use to see it in the preference menu . Even the latest Beta doesnt show it for me. Oh and Im on Linux

I definitely saw it on the AppImage build a few weeks back... It's not there currently though. It's in a weird situation right now. 

The AppImage build is in dark mode (no option for light mode) and the Flatpak version is in light mode (no option for dark mode).

cochcoder, 2025-03-19:
While #8964 doesn't add a toggle for dark/light mode, it should fix the issue that you experienced.

dannyglover, 2025-03-21:
> While [#8964](https://github.com/SoftFever/OrcaSlicer/pull/8964) doesn't add a toggle for dark/light mode, it should fix the issue that you experienced.

Hello. I tried (and failed) to get dark mode working with either the latest public release (2.3.0), or the latest nightly build (at the time of commenting).

Am I too early?

[image]

cochcoder, 2025-03-22:
You have to use the build from the PR. And you shouldn't need to add the ORCA_SLICER_DARK_THEME variable.

dannyglover, 2025-03-22:
> You have to use the build from the PR. And you shouldn't need to add the ORCA_SLICER_DARK_THEME variable.

Ah, gotcha.

I assume this'll be merged into main at some point?
Thanks again.

cochcoder, 2025-03-25:
> > You have to use the build from the PR. And you shouldn't need to add the ORCA_SLICER_DARK_THEME variable.
> 
> Ah, gotcha.
> 
> I assume this'll be merged into main at some point? Thanks again.

Hopefully. I did set my PR to fix this issue so as soon as it gets merged, this issue will be automatically closed.

closed 2025-03-30 by SoftFever

cochcoder, 2025-03-31:
> Still not working on v2.3.1-dev build [c163d11](https://github.com/SoftFever/OrcaSlicer/commit/c163d11fdb8cb781a08cf0c17d81c75c86d2b413). Using Flatpak version x86 on Linux Ubuntu MATE 24.04 LTS OrcaSlicer v2.3.0 stable AppImage works fine. Additionally, would be really nice to just have a toggle for this the same as in the Windows version.

If you run it with the `ORCA_SLICER_DARK_THEME=true` variable set. Does it lunch in dark mode?

MxBrnr, 2025-04-03:
I admittedly do not know how to set that; not that well versed in Linux systems. I know that the AppImage version by default runs in whatever mode the desktop environment is using (currently dark mode), but the FlatPak version does not, and is stuck in light mode, at least for me.

darco415, 2025-04-04:
Well hell, I changed my desktop theme from
dark to light and now OS is using  light theme, which I want cuz now I can see my plate but I want a dark theme desktop! This is latest 2.3 and no there is still no toggle switch for me. Linux appimage

dannyglover, 2025-04-06:
> If you run it with the `ORCA_SLICER_DARK_THEME=true` variable set. Does it lunch in dark mode?

Not the guy you asked, but I tried this again (setting the env var with flatseal) with the latest dev flatpak and it works as expected now.

MxBrnr, 2025-07-05:
Latest v2.3.1 build also does not work on vanilla Ubuntu 25.04. This issue is definitely _far_ from being solved.

dmayhem, 2025-07-05:
I did a v2.3.1 build on Fedora 42
Dark mode works fine as native executable
Dark mode works fine as appimage
Have not yet done a flatpak build. TBH, may not bother. I'm not a huge fan of flatpak vs appimage, and now that I have a working appimage on Fedora 🤷‍♂️

bmentink, 2025-08-16:
Doesn't help me. I am on Linux arm64, ther is no .appimage for that target, so I have to use Flatpak to run OrcaSlicer. I too want dark mode..

pm0u, 2025-08-20:
Confirm this works for me on 2.3.1-alpha with flatpak

ChristianJohnsten, 2025-09-06:
I'm using Orca Slicer 2.3.1-alpha flatpak and setting the `ORCA_SLICER_DARK_THEME=true` flag using flatseal is not changing the theme at all.
I am on Fedora 41

samin50, 2025-11-13:
I am getting this issue on 2.3.1 Build f2d10a0 (Flatpak) on Arch Linux/KDE Plasma 6.5.2. There is no dark/light checkbox in the Preferences, and setting `ORCA_SLICER_DARK_THEME=true ` does not work. I need to use the Flatpak version to get proper icon scaling.

rptorres22, 2026-01-03:
I was having this issue in Fedora 43 KDE. I came across this video and it solved the dark theme issue: https://www.youtube.com/watch?v=26frRGBqulU

In short:
1. Copy folder contents of /usr/share/themes/
2. Create a folder named ".themes" in /home/yourusername/
3. Paste the contents in /home/yourusername/.themes/

Hope this helps anyone until an official fix is in place.

Rippeee, 2026-03-09:
i had same problem in CachyOS with the latest Flatpak version 2.32-rc2.

I just added a new environment in Flatseal, and that fixed it. ( GTK_THEME=Adwaita:dark)

[image]

PR #8964 (2025-03-30): Fix flatpak not using dark theme
fixes #8518

This PR fixes the Orca Slicer flatpak not using the dark theme when launched on DEs outside of Gnome. These changes originate from [Prusa Slicer's flatpak](https://github.com/flathub/com.prusa3d.PrusaSlicer) with the only change being `ORCA_SLICER_DARK_THEME`, rather than `PRUSA_SLICER_DARK_THEME` and a mention to a change originating from Prusa Slicer which originated from Spotify.

Before:
[image]

After:
[image]

Tested on:
Fedora 41
KDE Plasma 6.3.3
Breeze Dark theme