OrcaSlicer/OrcaSlicer #8526: Bambu x1 lan only mode not functional

broncosis, 2025-02-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 -beta build 49116

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.2 Cinnamon  Kernel  6.8.0-52-generic

### Additional system information

CPU Ryzen 7 3700 32 gigs ram 

### Printer

Bambu x1c Lan only mode 

### How to reproduce

send a print to the printer 
it asks for the IP and access code for the printer then asks for the printers name and serial number and model 
after filling this out it attempts to send files and then opens the dialog to enter the info again 

Stealth mode on or off does not help 

### Actual results

the results are a endless loop of trying to put printer identification in 

### Expected results

Orca should save the information and upload the file to the printer
like it did in the previous version

 

### Project file & Debug log uploads

[failtoconnect.zp.zip](https://github.com/user-attachments/files/18928863/failtoconnect.zp.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

broncosis, 2025-02-23:
this apears to be a bug with the new network plugin I guess ass now I can't use the previous version of ORCA either

thojo0, 2025-02-23:
I have the same problem with a P1S.
For me auto discovery also doesn't work.
Also there is only the "connect by pin-code" option and not the direct IP option anymore.

[image]

So I can't even try to connect to the printer, but my Home Assistant still can connect.

spali, 2025-02-24:
Do you have any special setup? For me with my X1C it works after upgrading from 2.2.0 to 2.3.0-beta. It even remembers now the printer as soon you go to the "Device" tab. My Printer is in Lan only mode and blocked from the internet. Orca (my client) is allowed to connect to the internet. Printer is in the same VLAN as the client, so no special setup based on auto-discovery.

closed 2025-06-02 by github-actions[bot]