OrcaSlicer/OrcaSlicer #8533: Orcaslicer crashes after slicing, also puts purge tower off plate

W1SPS, 2025-02-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta Build 2bba98d

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

MSI MAG Tomahawk Z690 wifi MB
Intel i9 12900K
64GB DDR5
Nvidia RTX A4000 Quadro

### Printer

BBL X1C

### How to reproduce

Try to slice anything and when completed program crashes.

Also when printing with multiple filaments the purge tower often get put partially off of build plate.

### Actual results

Program just quits

### Expected results

Program should not quit

### Project file & Debug log uploads

[1 Complete Sets - 25 Cal - PETG w-support- X1C.zip](https://github.com/user-attachments/files/18930290/1.Complete.Sets.-.25.Cal.-.PETG.w-support-.X1C.zip)

[debug_Sun_Feb_23_07_06_53_27688.log.zip](https://github.com/user-attachments/files/18930296/debug_Sun_Feb_23_07_06_53_27688.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Calisto88, 2025-02-23:
Having same issues .

First this started to happen when i had tree auto supports, and then decided to change and set it for tree manual , marked the support areas and when i hit slice it just instant closes when it reaches generating supports .

The only way i had to stop it is to set it up for auto back again and then was slicing normal.

But now out of the sudden , i switched OFF slow down for overhangs and i cant slice it back anymore. i need to have the option ON .

W1SPS, 2025-02-23:
It seems to be getting worse. it intermittently crashes randomly when you interact with various parts of the program. BBS is doing the same crap. I've removed both and deleted their directories, rebooted then reinstalled. No help.

Calisto88, 2025-02-23:
Well i dont know at the moment with beta 2 for me still didnt got any crash !

W1SPS, 2025-02-24:
I'm hoping that the  log file and project file help. If this was being cause by, say, a conflict with a video driver would the log file indicate that?

Noisyfox, 2025-02-28:
Try change the cpu affinity so Orca runs on P cores only.

Noisyfox, 2025-02-28:
Do you have logs that file name start with crash_ ?

W1SPS, 2025-02-28:
There is definitely something wrong with how orcaslicer is handling multiple materials. Every time I add 1 extra material for filling in text on the surface of a part I get "**Error: WipeTowerIntegration::append_tcr was asked to do a toolchange it didn't expect.**" I've tried this on two completely different computers, I've tried 3 different materials instead of 2, I've tried re-STEP'ing the parts out, re-orient the parts, changed the depth of the lettering, nothing works. This problem started 2 weeks ago and NEVER happened before that in all the 3 years I've owned the X1C.

W1SPS, 2025-02-28:
> Do you have logs that file name start with crash_ ?

When I started this tread I uploaded the LOG and the PART that I am trying to process, clearly no one looked at it as neither has been mentioned since. There is something going on with the multimaterial handling that didn't exist before. It's almost like code from a multi-toolhead machine was introduced in the X1C code and orcaslicer is issuing a toolhead change instead of a filament change. But I couldn't say, I'm a designer not a programmer.

Noisyfox, 2025-02-28:
Did you use any post-process scripts?

W1SPS, 2025-02-28:
I don't know what that is, but since no one looked at what I did send I don't see how that would help.

Noisyfox, 2025-02-28:
> I don't know what that is, but since no one looked at what I did send I don't see how that would help.

I was asking for more info simply because I've looked at EVERYTHING you already posted and I couldn't reproduce any crash so far.

W1SPS, 2025-02-28:
Well if you take the project file, put it in your slicer, paint the text on the flat part with the lettering and arrow on it and slice it you should get the "Error: WipeTowerIntegration::append_tcr was asked to do a toolchange it didn't expect."  at the very least

W1SPS, 2025-02-28:
Here is some of the setting pictures: https://drive.google.com/drive/folders/1cTVI_pDlkzfBg8TnHgepMmjVgIYHSO5N?usp=drive_link

If you need/want to see anything else let me know

Noisyfox, 2025-02-28:
[image]
The letters are already painted and I could slice it normally. I also did some extra painting on it but still didn't see anything unusual.

[image]

W1SPS, 2025-02-28:
I don't get it. I get the wipetowerintegration thing every time on every computer I run it on

Noisyfox, 2025-02-28:
Do you mind uploading (or email me via timemanager.rick(at)gmail.com" the entire "<user>\AppData\Roaming\OrcaSlicer" folder ?

W1SPS, 2025-02-28:
Sent you a link via gmail

Noisyfox, 2025-03-01:
I'm even more confused now after inspecting all the crash logs. It crashed at different unrelated places each time, some of which even doesn't involve slicing process.

Noisyfox, 2025-03-01:
Though I did find one case of your crash was caused by the same reason as #8604 which should be fixed now in the latest nightly build.

closed 2025-06-07 by github-actions[bot]