OrcaSlicer/OrcaSlicer #8535: Asahi Linux Build Support

amdavidson, 2025-02-23:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

There is no current aarch64 builds of OrcaSlicer that will run on Asahi, but it does build on Asahi and adding a linux.d file would make it easier for Asahi users to get Orca running.

An working option for me: https://github.com/amdavidson/OrcaSlicer/blob/asahi/linux.d/fedora-asahi-remix

This is nearly the same as the standard fedora file with the removal of libquadmath, which is not packaged for Asahi. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add the relevant file to linux.d to allow for builds on Asahi.

### Describe alternatives you've considered

Prebuilt AppImages or Flatpak would be preferred, but the relevant github runners seemingly do not exist yet. 

### Additional context

These builds appear to be fully functional despite the removed dependency with one caveat, the Bambu network plugin is unable to load and gives and error and re-tries installation with every launch.

The error it gives me is: 
```
loading network module at /home/amd/.config/OrcaSlicer/plugins/libbambu_networking.so  
error, dlerror is /home/amd/.config/OrcaSlicer/plugins/libbambu_networking.so: cannot open shared object file: No su  
ch file or directory  
after dlopen, network_module is (nil)
```

If there is a flag to disable the plugin support (probably a good idea given its tenuous existence) that could be added to the `fedora-asahi-remix` file along with a warning in the console.

cdinerman, 2025-02-24:
This change worked great for me (MBPro 2021, Fedora Asahi Linux), thanks @amdavidson!

cochcoder, 2025-02-28:
While this doesn't affect the issue of compiling Orca Slicer on Asahi Linux, I have successfully made a arm64 flatpak build of Orca Slicer in PR #8033. Any tests would be greatly appreciated as the only arm computer I own is a Raspberry pi 3 b+ and I'm not sure how many bugs are from it being slow and having only 1GB of ram. The Flatpak file for arm can be found [here](https://github.com/SoftFever/OrcaSlicer/actions/runs/13580568050?pr=8033).

cdinerman, 2025-02-28:
@cochcoder The flatpak worked well. The only issue I noticed was that I cannot connect to my printer over the network (which works on my self-compiled version).

cochcoder, 2025-03-02:
> [@cochcoder](https://github.com/cochcoder) The flatpak worked well. The only issue I noticed was that I cannot connect to my printer over the network (which works on my self-compiled version).

What specific error did you get when you tried to upload to your printer? Also, what printer are you using?

cdinerman, 2025-03-04:
@cochcoder my printer is a voron running Klipper. 

The error I get when testing is:
`Could not connect to OctoPrint: curl: Could not resolve host: Could not resolve host: v24.local [Error 6]`

Similarly, the device tab says:
`Error resolving “v24.local”: Temporary failure in name resolution`

cochcoder, 2025-03-05:
@cdinerman 
I've tried downgrading the version of gnome used from 47 to 46. The new build can be found [here](https://github.com/SoftFever/OrcaSlicer/actions/runs/13677642051). (The build should be done around one hour and 30 minutes after this comment)

cdinerman, 2025-03-06:
Still getting the same errors on the new build.

cochcoder, 2025-03-06:
> @cochcoder Still getting the same errors on the new build. Everything else is working great, just the networking is being weird.

Do you have happen to have the BambuLabs network plugin installed? Another tester is able to connect to their printer as long as they don't have the network plugin installed.

cdinerman, 2025-03-06:
@cochcoder The BambuLabs network plugin was installed. I disabled it, made sure stealth mode was enabled, and restarted the application. Same network errors still. Thanks for continuing to research this!

cochcoder, 2025-03-06:
@cdinerman 
Could you try completely removing the flatpak, including any user data it saved, and reinstall the flatpak without the BambuLabs network plugin? I suspect that the plugin is potentially causing the issues, even though it's disabled.

cdinerman, 2025-03-06:
I just removed it, including the delete-data option and reinstalled the flatpak. I did the setup routine, rejected the bambulabs network plugin and activated stealth mode. 

Same error. (Sorry)

cochcoder, 2025-03-07:
@cdinerman 

I've started a new [build](https://github.com/SoftFever/OrcaSlicer/actions/runs/13722347690/job/38380569215?pr=8033) that should include the curl dependency in the flatpak, hopefully it work this time. 

Thank you again for continuing to test these changes!

cdinerman, 2025-03-09:
@cochcoder 
I removed the previous flatpak (and data) and installed the latest one. I still got the same errors as before.

So I decided to try something different and add it (v24.local) my hosts file. **That fixed it!**

I still don't understand why this is an issue in the flatpak and not the compiled program, but there is a work-around!

Thank you!!!!

cochcoder, 2025-03-09:
@cdinerman 
Thank you for finding that work around! Could you explain how you added Orca Slicer to your v24.local file more?

cdinerman, 2025-03-09:
@cochcoder 

It was a DNS resolution issue. 

1. I pinged my printer `ping v24.local`
2. noted the ip address (lets call it 192.168.1.123)
3. edited my hosts file `vi /etc/hosts` and appended `192.168.1.123 v24.local` to the file

cochcoder, 2025-03-09:
@cdinerman 
Interesting, hopefully this issue is localized to Asahi Linux and will be fixed soon. I'm going to go ahead and mark the PR as ready to merge as another person on a separate distro doesn't have this issue. Thank you again for testing the flatpak.

cdinerman, 2025-03-09:
@cochcoder 

Thanks for your hard work making Orca Slicer more broadly available!

amdavidson, 2025-03-09:
This is working for me on Asahi without any DNS issues. Thanks for the work @cochcoder

cochcoder, 2025-03-12:
@amdavidson 

I'm new to the compilation area, however I think that if you add `export DISABLE_BAMBU_NETWORK_PLUGIN=1` to the bottom of the Linux.d file, it should automatically disable the Bambu network plugin

genericapplesauce, 2025-03-17:
I can also confirm this seems to work on a Apple MacBook Pro (14-inch, M1 Pro, 2021) running Asahi Fedora Remix 40. I'll test it out more thoroughly and report if I run into anything, but thank you so much for your work on this - having a slicer in ARM Linux is really helpful for me!

amdavidson, 2025-12-30:
I no longer have an asahi machine to test with, but it sounds like it works. Someone can re-open if they can test further.

closed 2025-12-30 by amdavidson