OrcaSlicer/OrcaSlicer #8538: Export pre set bundle  closes the orca slicer software 2.3B

moth4017, 2025-02-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3B

### Operating System (OS)

Windows

### OS Version

windows 10 pro

### Additional system information

[image]

### Printer

Valkyrie stage 1

### How to reproduce

file
export
export pre set bundle

program exits  

### Actual results

program closes 

### Expected results

would expect to go to export bundle page 

### Project file & Debug log uploads

there is no project file 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

eseer3, 2025-02-27:
Using 2.3B... I am unable to export to a cloud drive but worked on local drive (desktop). win 10

Noisyfox, 2025-02-28:
Dup of #8510 and has been fixed in #8525

closed 2025-02-28 by Noisyfox