OrcaSlicer/OrcaSlicer #8540: Add Qidi i-Mates 0.4 nozzle printer profile

ageoffri, 2025-02-23:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

The Qidi i-Mates printer does not have a profile.  Adding the printer profile would be beneficial to a number of people as I have seen requests for this printer in both Reddit and Discord. 

### Which printers will be beneficial to this feature?

Marlin

### Describe the solution you'd like

I have created a 0.4mm nozzle printer profile for the Qidi i-Mates and have successfully printed multiple PLA models using it. The profile is based on the settings from Qidi Slicer, Cura, and direct queries to the printer using G-code commands. I also used ChatGPT to help understand the differences in setting names between various slicers and Orca.

Since I use OctoPrint, I haven't tested the profile with WiFi. To connect OctoPrint to the printer, I had to remove the WiFi module.

Although I haven't contributed to a project like this before, I do use GitLab with Terraform at work. I’ve created a fork and will be submitting a pull request. Any feedback would be greatly appreciated to refine this profile so others can benefit from it and potentially improve it further.

thelegendtubaguy, 2026-11-08:
I think this is likely a situation where Orca won't be adding it. Qidi doesn't provide profiles for this in QidiStudio, so it looks abandoned.  I would suggest closing in favor of running a custom profile.  With Orca cloud, these are easily shared.

raistlin7447, 2026-11-08:
Thanks for putting this together. Closing this as not planned.

Qidi doesn't ship an i-Mates profile in QIDIStudio, so there's no official profile to bring over. Your PR (#8541) was closed because it only had the machine file. For now a custom printer profile like the one you made is the way to go.

If Qidi publishes an official i-Mates profile, we'll reopen this.

closed 2026-11-08 by raistlin7447