OrcaSlicer/OrcaSlicer #8544: Adjust/assign the flow rate for each feature

SebKoAIM, 2025-02-24:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I am currently printing with a pellet printer that requires an adjusted extrusion multiplier (EM) depending on the speed (flow).

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Since different features are printed at different speeds, I am currently using the search and replace function with a simple python script in post-processing to change the override. Is there a possibility to introduce a feature like in Cura that allows setting the flow for individual features?

Thank you very much and best regards.

tiziano149, 2025-03-12:
Absolutely want this. This is needed for professional results.

iamnotahippy, 2025-04-12:
This has been requested multiple times, and is an absolutely essential feature.