OrcaSlicer/OrcaSlicer #8545: Some synced filaments are reported under the wrong manufacturer

lukasmalkmus, 2025-02-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

v2.3.0-beta (dc49116)

### Operating System (OS)

macOS

### OS Version

15.3.1

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Install v2.3.0 beta
2. Connect to Printer (LAN mode, different subnet)
3. Re-sync filaments

### Actual results

Some filaments are reported as being from AliZ instead of Bambu, for me that's notably Bambu PETG family filaments.

The AMS status is reported correctly:

[image]

But filament selection after re-sync is broken (PETG being reported as AliZ instead of Bambu manufactured):

[image]

I have also never checked AliZ filaments:

[image]

### Expected results

Correctly sync filaments.

### Project file & Debug log uploads

None

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

At this point, I can't share the project file until I've made sure some data has been removed / is not included.

AaronMickDee, 2025-02-24:
Can confirm. Also missing Generic PETG on Device tab for Bambu Lab X1C

spali, 2025-02-24:
Confirm the `Generic PETG` get's `AliZ PETG` and also `PolyTerra PLA` gets `PolyTerra Dual PLA`.
On the printer itself it shows correct, but just somehow maps wrong to Orca. But it's also not consitent in Orca itself. 
All my on the printer selected `PolyTerra PLA` shows as `PolyTerra Dual PLA` in the filament selection in `Prepare` for the AMS. But it's correct in `Device` when I click on the Filament in the AMS.
But `AliZ PETG` is wrong in both places.

pgifford, 2025-02-25:
Came here to report the same problem.
1. Select "PolyTerra PLA" in AMS (either on printer or in slicer)
2. In slicer click "Synchronize filament list from AMS"
3. Filament list shows "PolyTerra Dual PLA"

[image]

---

[image]

"PolyTerra Dual PLA" is unnecessary and should be removed. Base PolyTerra PLA settings are fine for their dual matte and dual gradient filaments.  Strange that PolyLite Dual silk wasn't added at the same time.  Regardless, all Polymaker filaments have been renamed anyway so eventually it will all change to the many flavors of Panchroma.

atl285, 2025-02-25:
For me `Generic PLA` and `Generic PETG` get synced to `AliZ PLA` and `AliZ PETG`. None of the `AliZ` filaments is selected as available system filament. It seems, that for `Generic` filaments the **first** filament vendor is selected.

lukasmalkmus, 2025-02-25:
> It seems, that for `Generic` filaments the **first** filament vendor is selected.

Yeah, I thought the same. Previously `Bambu` was always the first available vendor as the list is alphabetically ordered so with new vendors added, there seems to be a mapping problem where it automatically assigns the first vendor from that list.

spali, 2025-02-25:
Looks like duplicate of https://github.com/SoftFever/OrcaSlicer/issues/8499 https://github.com/SoftFever/OrcaSlicer/issues/8509 and fixed in https://github.com/SoftFever/OrcaSlicer/pull/8531.
@SoftFever can you confirm it's the same we talk about here and your fixes in https://github.com/SoftFever/OrcaSlicer/pull/8531?

SoftFever, 2025-02-25:
> Looks like duplicate of https://github.com/SoftFever/OrcaSlicer/issues/8499 https://github.com/SoftFever/OrcaSlicer/issues/8509 and fixed in https://github.com/SoftFever/OrcaSlicer/pull/8531. [...]

Thanks @spali 
Yes, they are duplicated and the issue has been fixed.

lukasmalkmus, 2025-02-26:
Cool. Sorry for the noise. Apparently I didn't search closed issues properly. Sorry for the noice! Closing.

closed 2025-02-26 by lukasmalkmus