OrcaSlicer/OrcaSlicer #8548: Moving the changelog (update notification) window to a screen edges is laggy (causes a 100% CPU load)

Felix14-v2, 2025-02-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10/11

### Additional system information

CPU: i5 12400
GPU: GTX 1080 Ti
RAM: 16 GB 2667 MT/s

### Printer

N/a

### How to reproduce

1. Wait for an update
2. Open OrcaSlicer
3. Try moving the notification window as shown in the video
4. See the lags

### Actual results

https://github.com/user-attachments/assets/b86d9d6e-294e-4338-9dae-553da8496fd2

### Expected results

Optimized and smooth experience

### Project file & Debug log uploads

N/a

### Checklist of files to include

- [x] Log file
- [x] Project file

dstulken, 2025-02-26:
I tried this with OrcaSlicer 2.2.0 on macOS 12.7.6, and couldn't reproduce the high CPU usage... so it might be Windows-specific?

Felix14-v2, 2025-02-26:
I wouldn't be surprised. Also your CPU may be powerful enough to handle this load.

Felix14-v2, 2025-06-11:
> on OrcaSlicer 2.3.0 and couldn´t reproduce this issue

How did you manage to check the changelog for a new version on 2.3.0? 2.3.0 is already the newest available version.

MartinNovan, 2025-06-11:
You can slip, in the orcaslicer code, your URL link so that instead of the official repository you link to another repository (e.g. your own) where you can create your “fake” releases and make the changelog window pop up. You can also do this via DNS on your router, where you set where to link to when you request the orcaslicer repository. But unless you know C++ or networking I'd rather wait until the next release, hopefully it might be soon :D. Do you only have a problem with the changelog window or does this happen with other windows as well? Other users don't seem to have this problem. :/

Felix14-v2, 2025-08-22:
> Tried to reproduce this on Win 11 24H2 with i5-1335U + UHD graphics and 16GB RAM on OrcaSlicer 2.3.0 and couldn´t reproduce this issue

I'm still able to reproduce this in Orca 2.3.0:

https://github.com/user-attachments/assets/d4832717-1ca7-4c2c-a53b-b6637062aa2b

Windows 11 24H2

Notice how the text inside the window is cropped for a couple of frames when the window returns to the screen. It looks like some kind of text rendering optimization can't handle such a large changelog. And I assume that this is the same reason that leads to the slow scrolling bug (that occurs throughout the UI).