OrcaSlicer/OrcaSlicer #8550: Orca 2.2.0 Crashes on start on MacOS 15.3

sealyons, 2025-02-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

15.3

### Additional system information

M1 Max

### Printer

FlashForge 5M Pro 

### How to reproduce

2.2.0 crashes when started. Displaysstart screen, waits a few seconds then crashes.

### Actual results

Crashes every time I try to start it. It sometimes generates a crash report. Can't see any useful information in the report that indicates why it crashed.

### Expected results

Or a should load without crashing.

### Project file & Debug log uploads

There is no files to load. Orca will not start.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Trying to determine what the problem is. I: been unable to start Orca now for 2+ weeks. I did updated MacOS from 15.0.1 to 15.3 and It said run for a couple days. But then it started crashing again and hasn't started successfully since. Please help.

sealyons, 2025-02-24:
I have asked about this on the Discord channel but haven't received any answers.

Noisyfox, 2025-02-27:
Could you provide the crash report from the Console app?

sealyons, 2025-02-27:
Here's the contents of the Crash Report: (not sure how to attach the actual report, it's an .ips file and it's not allowing me to add that file type)

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OrcaSlicer [54652]
Path:                  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
Identifier:            com.softfever3d.orca-slicer
Version:               2.2.0 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-02-22 06:42:04.4710 -0500
OS Version:            macOS 15.3.1 (24D70)
Report Version:        12
Anonymous UUID:        CEBC05FF-9AFC-1AA6-07B3-A38FABF11120

Sleep/Wake UUID:       E3663BEA-E0CD-44D8-9EDB-3C5C0DA8CF55

Time Awake Since Boot: 620000 seconds
Time Since Wake:       107704 seconds

System Integrity Protection: enabled

Crashed Thread:        0  orcaslicer_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000007e38
Exception Codes:       0x0000000000000001, 0x0000000000007e38

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [54652]

VM Region Info: 0x7e38 is not in any region.  Bytes before following region: 4300931528
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1005b8000-104624000    [ 64.4M] r-x/r-x SM=COW  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer

Thread 0 Crashed:: orcaslicer_main Dispatch queue: com.apple.main-thread
0   OrcaSlicer                    	       0x10151ef30 Slic3r::GUI::Plater::get_notification_manager() + 4
1   OrcaSlicer                    	       0x1012b4500 Slic3r::GUI::GLCanvas3D::_set_warning_notification(Slic3r::GUI::GLCanvas3D::EWarning, bool) + 992
2   OrcaSlicer                    	       0x1012b2ba8 Slic3r::GUI::GLCanvas3D::~GLCanvas3D() + 168
3   OrcaSlicer                    	       0x10146a4d0 Slic3r::GUI::View3D::~View3D() + 40
4   OrcaSlicer                    	       0x1036f11ac wxWindowBase::DestroyChildren() + 44
5   OrcaSlicer                    	       0x103798330 wxWindow::~wxWindow() + 208
6   OrcaSlicer                    	       0x1036bd024 wxPanel::~wxPanel() + 20
7   OrcaSlicer                    	       0x1036f11ac wxWindowBase::DestroyChildren() + 44
8   OrcaSlicer                    	       0x103798330 wxWindow::~wxWindow() + 208
9   OrcaSlicer                    	       0x101565424 Slic3r::GUI::Plater::~Plater() + 12
10  OrcaSlicer                    	       0x1036f11ac wxWindowBase::DestroyChildren() + 44
11  OrcaSlicer                    	       0x103798330 wxWindow::~wxWindow() + 208
12  OrcaSlicer                    	       0x1014d92ec Notebook::~Notebook() + 124
13  OrcaSlicer                    	       0x1036f11ac wxWindowBase::DestroyChildren() + 44
14  OrcaSlicer                    	       0x103793ecc wxNonOwnedWindow::~wxNonOwnedWindow() + 92
15  OrcaSlicer                    	       0x1014d8aa4 Slic3r::GUI::MainFrame::~MainFrame() + 12
16  OrcaSlicer                    	       0x1036332c8 wxAppBase::CleanUp() + 84
17  OrcaSlicer                    	       0x1037c9584 wxApp::CleanUp() + 56
18  OrcaSlicer                    	       0x10388126c wxEntryCleanup() + 68
19  OrcaSlicer                    	       0x103881444 wxEntry(int&, wchar_t**) + 244
20  OrcaSlicer                    	       0x1014692b8 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 456
21  OrcaSlicer                    	       0x1005c9444 Slic3r::CLI::run(int, char**) + 14040
22  OrcaSlicer                    	       0x100627a0c main + 172
23  dyld                          	       0x19dca8274 start + 2840

Thread 1:
0   libsystem_pthread.dylib       	       0x19e0230e8 start_wqthread + 0

[... 509 lines cut ...]
  }
},
  "logWritingSignature" : "e94ef604039c63627f9d2a71ad737a094865f5e1",
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "644114de41e7236e6177f9bd",
      "factorPackIds" : {

      },
      "deploymentId" : 240000013
    },
    {
      "rolloutId" : "658361a603dc58766bab2d2a",
      "factorPackIds" : {
        "SIRI_TTS_DEVICE_TRAINING" : "6736733a26c1270f5a9ef53d"
      },
      "deploymentId" : 240000045
    }
  ],
  "experiments" : [

  ]
}
}

sealyons, 2025-02-28:
Anything I can try to get Orca running again on my Mac? Appreciate any guidance. Thanks.

Noisyfox, 2025-02-28:
Maybe try deleting the config files at ~/Library/Application Support/Orcaslicer.

sealyons, 2025-02-28:
What do those config files do? Will I lose any of my settings? Can I backup
those files first and copy back after trying this?

Noisyfox, 2025-02-28:
> What do those config files do? Will I lose any of my settings? Can I backup
> those files first and copy back after trying this?
> […](#)

Yes it's recomanded to backup the entire folder before doing anything to it.

sealyons, 2025-03-04:
Still can't start Orca 2.2.0 on my MacBook. It's been a few days and nothing. Please respond so I can get my printer running again.

PossibleCause, 2025-03-04:
I have same sort issue. Running mac 15.3 and orcasliscer  2.3.beta 2. The program crashes when after enetring the access code.
Here is the crash report:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OrcaSlicer [69325]
Path:                  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
Identifier:            com.softfever3d.orca-slicer
Version:               2.3.0-beta2 ()
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-03-04 12:38:52.2966 +0100
OS Version:            macOS 15.3.1 (24D70)
Report Version:        12
Bridge OS Version:     9.3 (22P3051)
Anonymous UUID:        6461947B-95C1-2BB1-DE5C-6C8634DCA41F

Sleep/Wake UUID:       5E53F0BF-6BE0-4649-BFA7-2808423B091A

Time Awake Since Boot: 190000 seconds
Time Since Wake:       2503 seconds

System Integrity Protection: enabled

Crashed Thread:        0  orcaslicer_main

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000058
Exception Codes:       0x0000000000000001, 0x0000000000000058

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [69325]

VM Region Info: 0x58 is not in any region.  Bytes before following region: 4462628776
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      109fe5000-10e820000    [ 72.2M] r-x/r-x SM=COW  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed:: orcaslicer_main
0   OrcaSlicer                    	       0x10a78cab0 0x109fe5000 + 8026800
1   OrcaSlicer                    	       0x10b5c8c95 0x109fe5000 + 22953109
2   OrcaSlicer                    	       0x10b164174 0x109fe5000 + 18346356
3   OrcaSlicer                    	       0x10da8c918 0x109fe5000 + 61503768
4   OrcaSlicer                    	       0x10da8c7b7 0x109fe5000 + 61503415
5   OrcaSlicer                    	       0x10da8c65f 0x109fe5000 + 61503071
6   OrcaSlicer                    	       0x10d9e8f77 0x109fe5000 + 60833655
7   OrcaSlicer                    	       0x10da96b31 0x109fe5000 + 61545265
8   CoreFoundation                	    0x7ff806d46173 0x7ff806ccb000 + 504179
9   CoreFoundation                	    0x7ff806d4608e 0x7ff806ccb000 + 503950
10  CoreFoundation                	    0x7ff806d45627 0x7ff806ccb000 + 501287
11  CoreFoundation                	    0x7ff806d44c6e 0x7ff806ccb000 + 498798
12  HIToolbox                     	    0x7ff81254c413 0x7ff81245d000 + 979987
13  HIToolbox                     	    0x7ff812551c2e 0x7ff81245d000 + 1002542
14  HIToolbox                     	    0x7ff812552029 0x7ff81245d000 + 1003561
15  AppKit                        	    0x7ff80a6a7b33 0x7ff80a667000 + 265011
16  AppKit                        	    0x7ff80b0fc8b8 0x7ff80a667000 + 11098296
17  AppKit                        	    0x7ff80a698aa9 0x7ff80a667000 + 203433
18  OrcaSlicer                    	       0x10d9b053e 0x109fe5000 + 60601662
19  OrcaSlicer                    	       0x10da9717d 0x109fe5000 + 61546877
20  OrcaSlicer                    	       0x10da11027 0x109fe5000 + 60997671
21  OrcaSlicer                    	       0x10d9e87c4 0x109fe5000 + 60831684
22  OrcaSlicer                    	       0x10d96ff7a 0x109fe5000 + 60338042
[... 485 lines cut ...]
}
}

Model: MacBookPro16,1, BootROM 2069.80.3.0.0 (iBridge: 22.16.13051.0.0,0), 6 processors, 6-Core Intel Core i7, 2,6 GHz, 16 GB, SMC 
Graphics: Intel UHD Graphics 630, Intel UHD Graphics 630, Built-In
Graphics: AMD Radeon Pro 5300M, AMD Radeon Pro 5300M, PCIe, 4 GB
Display: Color LCD, 3072 x 1920 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, DDR4, 2667 MHz, Micron, 8ATF1G64HZ-2G6E1
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, DDR4, 2667 MHz, Micron, 8ATF1G64HZ-2G6E1
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Jul 26 2024 22:36:01 version 9.30.514.0.32.5.94 FWID 01-68d7ff80
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: T2Bus
USB Device: composite_device
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5

PossibleCause, 2025-03-04:
My printer is a  Bambulab 1PS + AMS with firmware 01.07.00.00

KonkeBonk, 2025-04-04:
I have the same problem which has been bugging me for a couple of months now. I downloaded the latest version (2.3.0.) and the slicer crashes after I login with my Bambu account.

sealyons, 2025-07-05:
> Same Problem here

I have fixed this problem. You need to rebuild the configuration files,
MacOS can corrupt them periodically and it forces Orca to not load
properly. I'm away from my computer at the moment so I'd recommend going to
ChatGPT and giving it a prompt to suggest solutions for this problem.
That's how I got all the info I need to resolve it. Once I got the config
files rebuilt Orca has been running fine since on my MacBook since (no
issues).

Chris

BlackHawk3000, 2025-07-06:
I tried 2.3.1 night build and Orca is not crashing at the start. 
But i cant control the printer over cloud service, i dont know if its a problem with the Orca or with Bambu. 

I dont found any help with ChatGPT, only that i should delete or rename the Orca folder, after one start and login into the account, same problem.

sealyons, 2025-07-06:
2.3.1 is an experimental build. I wouldn’t use that unless you read through all the documentation and determine that it is solving some issue you have that the current stable build 2.3.0 does not. Personally, I never use the experimental versions, too many unknowns. I would recommend going back to the latest stable version, 2.3.0 and install that to be sure MacOS runs it correctly. If it does, then you understand what needs to be done if/when MacOS hiccups again later. 

Just to be clear, here are the exact steps I originally used to fix the Orca crashes:

1. In Finder, go to ~/Library/Application Support/OrcaSlicer and rename this folder to ~/Library/Application Support/OrcaSlicer_backup. This will backup all your current version settings if needed later.

2. Uninstall your current Orca version.

3. Download and install the latest stable Orca version. At this point it’s 2.3.0.

4. Go to ~/Library/Application Support/OrcaSlicer again (this was created again on the install). Rename this again to ~/Library/Application Support/OrcaSlicer_backupNEW. Now when you start Orca it won’t find the needed configuration files and it will rebuild the directory. 

5. With the configuration files rebuilt, Orca should now launch successfully. Start Orca and verify that it runs properly now.

6. If you need access to any profiles you had created, you can now go to your original configuration file backup,(~/Library/Application Support/OrcaSlicer_backup, and rename that back to ~/Library/Application Support/OrcaSlicer to allow access. If you do this, first rename the current folder to ~/Library/Application Support/OrcaSlicer_CurrentRunning so you secure your running instance. Note that it’s not a bad idea to do this anyway after install so you have a clean configuration file backup in case MacOS misbehaves again later. You should be able to go back here and delete the current ~/Library/Application Support/OrcaSlicer and rename this backup to that name and restart Orca and it should run properly.

If you need access to your old profiles, I’d recommend exporting them one by one and then going back to the known stable configuration folder and import them in one by one. This can be a pain if you have a lot of profiles, but it’s worth the work.

This should be all the needed steps to get Orca on MacOS functional again. Let me know if you have any questions. Good luck.

closed 2025-11-23 by github-actions[bot]