OrcaSlicer/OrcaSlicer #8556: Filament profile association

Help3D-Padova, 2025-02-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

Bambu Lab X1C

### How to reproduce

I have different 3D printers, not only Bambu Lab

1: Go to "Set Filament To use"
2: Select only Bambu Lab printers and Bambu filaments and add them
3: Make the same thing but with another printer with other filaments

### Actual results

1: Bambu printers get correctly associated with Bambu filaments and **only those** are visibile

[image]

2: Other printers have the selected filaments **AND** the Bambu ones + from other vendors

[image]

### Expected results

If I add a filament brand under a printer it should be visibile only there, like the Bambu behaviour in step 1.

Only with dependecies I can eventually share accross different printers.

Is a bug or is something related to my config ? I also tried to delete everything custom but nothing changes

### Project file & Debug log uploads

No Project file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Help3D-Padova, 2025-02-25:
> Found a workaround for you:
> 
> 1. go to your program installation folder. for me its C:\Program Files\OrcaSlicer\resources\profiles
> 2. Open OrcaFilamentLibrary.json
> 3. Search for "Bambu PLA Basic [@System](https://github.com/System)" and "Bambu PLA Matte [@System](https://github.com/System)"
> 4. Delete these lines.
> 5. Re-open your slicer.
> 
> This wont affect your profiles under Bambu printers. The only down side is that "Bambu PLA Basic [@System](https://github.com/System)" and "Bambu PLA Matte [@System](https://github.com/System)" wont be available for other printers.

Thank you for the reply

This is want is meant to be ? On the previous version of Orca Bambu Filaments were only on Bambu printers

Schildkroet, 2025-02-25:
This is a new feature
https://github.com/SoftFever/OrcaSlicer/releases/tag/v2.3.0-beta

SoftFever, 2025-03-03:
Yeah, I noticed this issue as well.
A per model filament selection is planned

Help3D-Padova, 2025-03-09:
@SoftFever thank you for the reply. In some cases is usefull, but in my (and many others) case i have so many different printers that needs different setting for fans, max volumetric flow, speeds ecc ecc . 

The approach similar to Prusa Slicer with dependancies is great, no need to share accross all printers the same system profile

#8779 i think is great

staal54a, 2025-03-17:
So did #8779 only address the issue with adding new filament profiles? I've noticed an issue with importing printer profiles that I had established with early versions of orca up to 2.2.0 and they were properly designated to my various printers. However, when upgrading to V2.3.0-rc (which I think includes that merge) I find that one of my printer profiles, my custom X1C profile, can see my user defined filament profiles for my X1C and one of my other printers (but not vice versa).

I can go in and manually update them with the new filament dependencies tab but that takes some time and while I don't have that many filament profiles, it's enough to keep me on V2.2.0 for now. It would probably be a lot more time consuming for people with even more profiles if they hit the same issue.

Just wondering if this is a known issue/accepted workaround? I have tried exporting my profiles in 2.2.0 and re-importing them in 2.3.0-rc (seemed to show promise since each printer bundle export only seems to have references to its tuned filament profiles via a quick notepad examination) but that didn't work either. Is there a way to edit multiple filament profiles at once with certain settings/dependencies? If so, I could give that a try.

closed 2025-07-17 by github-actions[bot]