OrcaSlicer/OrcaSlicer #8557: Incorrect bridges depending on the number of perimeters

Alexx40in, 2025-02-25:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambulab X1

### How to reproduce

The error is reproduced when slicing attached project file

### Actual results

If the number of walls in the settings exceeds the thickness of the walls of the part, the bridges are built incorrectly

### Expected results

The construction of bridges should not depend on the number of walls in the settings

### Project file & Debug log uploads

[image]

[Bridges.zip](https://github.com/user-attachments/files/18959656/Bridges.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Alexx40in, 2025-02-25:
> Walls are prioritised over bridges. This has been consistent throughout all the releases. If you would like to reduce the number of walls on a specific layer to enable a larger bridge please use a height range modifier :)

Special modes such as "sacrificial layer" or "partially bridged" do not depend on the number of walls in the settings

igiannakas, 2025-02-26:
There is a misconception as to how bridges are defined in orca/ prusa/ Bambu etc.

A bridge is composed of two parts:
1. Bridge perimeters
2. bridge infill

The bridge perimeters are technically perimeters, not bridge infill. They are printed with either the lowest overhang speed or the external bridge speed if detection for curled perimeters is disabled

The bridge infill is always printed with the external bridge speed, flow and spacing. This is shown as light blue in the slicer.

The implemented feature creates additional bridge infill regions and applied bridge logic (speed, spacing, flow) to them. It does not touch the perimeters.

In terms of print priority the perimeters always take priority in terms of space. Then any left over space is printed with bridge infill. Hence why narrow bridges don’t have any bridge infill - they are all technically overhang perimeters.

To optimise your print quality for narrow bridges the best way is to disable detection of curled overhangs for the object - this will force bridge speed on the bridge perimeters. Then if the bridge area is too narrow and flimsy, use a modifier to reduce the number of perimeters in that region. Lower number of perimeters also affects the space the bridge have for anchoring too.

In my view this is fundamentally how all the slic3r derivatives work. Learning to operate within their constraints with the tools available is a must - think bridges - use modifiers to reduce number of walls, disable curled overhangs for the object / region.

The above new feature works exceptionally for large bridges, which tend to be the ones where the bridge infill tends to get destroyed by the zig zag motion from the solid infill layer above.  Too narrow bridges only extrude perimeters hence they don’t suffer from this issue to that extent - as the layer over the bridge extrudes perimeters parallel to the walls. 

You can see some examples of the improvement areas in the Pr

lhndo, 2025-02-27:
> Special modes such as "sacrificial layer" or "partially bridged" do not depend on the number of walls in the settings

Yeah, that feature is quite amazing and similar to the SuperSlicer "No perimeter on bridge area" options. 
I just wish there was a way to specify the minimum amount of perimeters to keep.